jgarff/rpi_ws281x #233: What am I doing wrong? (any help is welcome)

Vioxe, 2017-10-11:
Hey everyone,
I don't know too much about how things work in the lunix world but I am trying to get a 300 digital led strip to run through the strandtest.py and it will not work.

Hardware I am using with the project: Raspberry Pi 2 Model B, 300 led strip 5 Meters WS2812b, a 5v 20 amp power supply, breadboard, jumper wires connecting grounds

The setup I currently have is that I have my V+ on my PSU running straight to my LED strip on the 5v wire, I have my V- running to my breadboard where I take two jumper wires from that, 1 going to the RPI on pin 6 which is ground and the other jumper wire is going to the LED strip on the ground wire. Then I have a wire going from my RPI on pin 12 or GPIO 18 to the data in on my LED strip. I am running Rasbian stretch lite newest version to date. I have blacklisted audio drivers, forced audio to hdmi, basically everything I could think of to get this strip working with my RPI and am now going to the forums for help.

Thanks in advance for any help you can supply me with.

siggy, 2017-10-11:
Do you have a level converter between the RPI GPIO and the data pin on the LED strip? The GPIO on the Pi is 3.3V, you need to convert to 5V.

More details:
https://learn.adafruit.com/neopixels-on-raspberry-pi/wiring#level-converter-chip-wiring

penfold42, 2017-10-11:
Try changing the dma channel to 10 instead of the default which is 5

Vioxe, 2017-10-11:
Thanks for the quick response and no I do not have a level converter. Could you link me one on Amazon that would be sufficient. I will try changing the dma channel to 10 when I get home.
Thanks for the help!

Vioxe, 2017-10-12:
I tried the dma change and I got nothing different in my results

penfold42, 2017-10-12:
What are your symptoms ? 
No lights ? Flickering ?

My favourite level shifter is just about any 74HCT series chip. 74HCT125 is popular.

Vioxe, 2017-10-12:
Yup no lights or flickering. About 2 weeks ago when I got all my items for the project I tried the strandtest.py with same hardware and software and the lights flickered and the did the animations but as I said there was flickering in the lights. I then put a fresh install on the RPI and then it no longer worked at all

Vioxe, 2017-10-12:
I will try the level converter that you suggested

sschroe, 2017-10-14:
Just a few more suggestions, although not necessarily directly related to your current issue.

* Don't connect the LED strip V- through a jumper wire. Hook it directly to the power supply. The thin jumper wires aren't really made to handle 18 amps (unless you have some special ones maybe).
* Feed power into both ends of the strip and use thick wires to connect to the power supply, this prevents the LEDs on the far end appearing darker due to voltage drops.

But as already suggested your initial problem more likely is related to the missing level shifter.

Gadgetoid, 2018-03-06:
I'm fairly confident that the conclusions above are correct: that the lack of a level shifter in this setup is quite likely the cause of the issue. I'll close this for now, but feel free to re-open if you solve your electrical problems and continue to have issues with the code.

closed 2018-03-06 by Gadgetoid

Vioxe, 2018-07-24:
I got the 74AHCT125 and i have the wiring done and perfected but still no lights or flickering. Could the issue be that i have the Raspi 2 model B, and that this library doesnt support Raspi 2 model B? Any help diagnosing my continued issues would be greatly appreciated. Thanks!

Vioxe, 2018-07-24:
https://drive.google.com/drive/folders/1UXDPYXyKkWudv3kN7SAvm4OLvB0IC06K?usp=sharing
There's my wiring for it all.

Vioxe, 2018-07-24:
I have the library installed and when i go to run strandtest.py it runs and goes through printing out the animations but nothing happens on the light strip.