jgarff/rpi_ws281x #234: Unable to build with CGO (scons works fine)

bmhatfield, 2017-10-19:
On Go 1.9.1, `go build` of this file:

```
package main

import (
	"github.com/jgarff/rpi_ws281x"
)

func main() {
	fmt.Println("hello world")
}
```

(the import is possible due to copying the files out of `golang/ws2811`)

results in the following error:

```
# github.com/jgarff/rpi_ws281x
mailbox.c: In function ‘mapmem’:
mailbox.c:47:33: warning: implicit declaration of function ‘getpagesize’ [-Wimplicit-function-declaration]
     uint32_t pagemask = ~0UL ^ (getpagesize() - 1);
                                 ^
mailbox.c: In function ‘mbox_open’:
mailbox.c:275:9: warning: implicit declaration of function ‘mknod’ [-Wimplicit-function-declaration]
     if (mknod(filename, S_IFCHR|0600, makedev(100, 0)) < 0) {
         ^
mailbox.c:275:25: error: ‘S_IFCHR’ undeclared (first use in this function)
     if (mknod(filename, S_IFCHR|0600, makedev(100, 0)) < 0) {
                         ^
mailbox.c:275:25: note: each undeclared identifier is reported only once for each function it appears in
```

I have a hunch that there's some flag that isn't being handled by the cgo build. The test app builds fine with scons and works on my machine (rpi 3B).

bmhatfield, 2017-10-19:
The flags in ws2811.go are:

```
// #cgo CFLAGS: -std=c99
// #cgo LDFLAGS: -lws2811
// #include "ws2811.go.h"
```

bmhatfield, 2017-10-19:
Some StackOverflowing suggested that those values were preprocessed out. Adding `-D_GNU_SOURCE` to the flags (ie; `// #cgo CFLAGS: -std=c99 -D_GNU_SOURCE`) gets past that stage and on to a linking error:

```
# github.com/jgarff/rpi_ws281x
/tmp/go-build786221241/github.com/jgarff/rpi_ws281x/_obj/main.o:(.data+0x70): multiple definition of `ledstring'
/tmp/go-build786221241/github.com/jgarff/rpi_ws281x/_obj/ws2811.cgo2.o:(.data+0x0): first defined here
/tmp/go-build786221241/github.com/jgarff/rpi_ws281x/_obj/main.o: In function `main':
../../jgarff/rpi_ws281x/main.c:376: multiple definition of `main'
/tmp/go-build786221241/github.com/jgarff/rpi_ws281x/_obj/_cgo_main.o:/tmp/go-build/github.com/jgarff/rpi_ws281x/_obj/_cgo_main.c:1: first defined here
/usr/bin/ld: cannot find -lws2811
collect2: error: ld returned 1 exit status
```

bmhatfield, 2017-10-19:
SConstruct seems to suggest that `-lws2811` is outdated - it only references `-lrt`.

Replacing that results in:

```
# github.com/jgarff/rpi_ws281x
/tmp/go-build204925519/github.com/jgarff/rpi_ws281x/_obj/main.o:(.data+0x70): multiple definition of `ledstring'
/tmp/go-build204925519/github.com/jgarff/rpi_ws281x/_obj/ws2811.cgo2.o:(.data+0x0): first defined here
/tmp/go-build204925519/github.com/jgarff/rpi_ws281x/_obj/main.o: In function `main':
../../jgarff/rpi_ws281x/main.c:376: multiple definition of `main'
/tmp/go-build204925519/github.com/jgarff/rpi_ws281x/_obj/_cgo_main.o:/tmp/go-build/github.com/jgarff/rpi_ws281x/_obj/_cgo_main.c:1: first defined here
collect2: error: ld returned 1 exit status
```

Doing `rm main.c` seems to remove the error and allow the build to complete.

bmhatfield, 2017-10-19:
I'd be happy to submit a PR to shore these things up, but I'm actually not really sure what I have done here is correct or appropriate.

Hopefully my comment log helps someone else, or helps @jgarff update the `go` files.

bmhatfield, 2017-10-19:
Confirmed that the Go code works properly* after the above notes.

* properly except that the RGB color values don't seem to map correctly. Perhaps the STRIP_TYPE def is not correct for the Alitove strip I am using. Still researching how to fix that.

KevinNovak, 2018-04-17:
#288