jgarff/rpi_ws281x #235: How to manipulate LED Colors

jpendress, 2017-10-26:
The Readme says that it is easy to set Colors etc for each LED by manipulating the LED Array, but it dont say how.

In one exampe i found this code 👍 

`  long b;
  for (b = 0; b < 99 ; b++) {
    int i;
    long x=(b*65536l)+(b*256l)+b;
    for (i = 0; i < LED_COUNT ; i++) {
      ledstring.channel[0].leds[i] = x;
    }`

The first value in the for should be : 657923 but what does the value mean ?! 
Can somebody please give me an example how to maniupulate the LED String ?

jpendress, 2017-10-29:
Can nobody answer my question ? I cant find any hints about programming this lib in c++
 will try it in Python now, but i dont like it. I would prefer c++ :(

dushkostanoeski, 2017-10-29:
Look at it like this: `0x00RRGGBB` (depending on the strip you have, the order of the colors might differ). 
Also, the higher the number, the brighter the color, so `0x00200000` will be a dim red and `0x00FF0000` will be bright red.

jpendress, 2017-10-30:
Thanks alot ! That sounds logical ;) 
Is there a hint in the sourcecode or Readme ? I didnt find anything...

dushkostanoeski, 2017-10-30:
I have a bit of a design background and when I saw this part of the code: 
```
ws2811_led_t dotcolors[] =
{
    0x00200000,  // red
    0x00201000,  // orange
    0x00202000,  // yellow
    0x00002000,  // green
    0x00002020,  // lightblue
    0x00000020,  // blue
    0x00100010,  // purple
    0x00200010,  // pink
};
```
it dawned to me how it works :)

jpendress, 2017-10-30:
I saw this part of the code, that should have been enough "hints" :D
Thanks alot !

dushkostanoeski, 2017-10-30:
No probs :) Just don't forget to close the issue

closed 2017-10-30 by jpendress