jgarff/rpi_ws281x #236: Slow display frequency - WS2812B

laszlocsk, 2017-10-30:
I'm having difficulties with the refresh rate of a WS2812B strip.
I have 15m LED strip with 300 pixels all together connected to the SPI MOSI pin. They work fine, but my problem is when I have only 5m of the LEDs addressed it works significantly faster than using all of them. Having all of the 15m in use I get the same results whether I have `time.sleep(50 / 1000)` or `time.sleep(500 / 1000)` in my code.

sschroe, 2017-10-30:
Each single led requires a fixed amount of time (depending on the used frequency), thus the more leds you add the longer the refresh is going to take. Using just 5m of the strip would mean only 1/3 of the refresh time is necessary.

For 300 leds it would take ~9ms per cycle at maximum speed. Though i'm not sure if the SPI code runs at the max frequency. This is purely for driving the output signal, additional delays might be added from your own code.

laszlocsk, 2017-11-10:
I have the following code:
`def Test:
    for i in range(strip.numPixels()):
        strip.setPixelColor(i, Color(0, 0, 255))
        strip.show()
`
and it takes 57.75 secs to run, so my question is if this can be lowered in any way.

penfold42, 2017-11-11:
Take strip.show out of the loop

laszlocsk, 2017-11-12:
What would be the point of having a digital LED strip if I can't run animations which update one LED at a time? Also I ran the code so that I know how fast the LEDs are and right now they can only run with ~15fps with that real simple code I mentioned before.

sschroe, 2017-11-12:
You can run animations just fine by having the strip.show() outside of the loop. What strip.show() does is to push the colors for all LED's in the strip down the wire. So currently you change a single color and then update the entire strip before going to the next color. Instead you should set the colors that you want to be shown for the current animation frame and only after that call strip.show() to push the color data to the LED's.

laszlocsk, 2017-11-14:
There's clearly a big difference how it'll look like whether you push the data every cycle or just at the end.

Note again that the aforementioned code's purpose is only to see how fast can the LEDs run with this library and not to color every the entire strip to blue. While it's true that the LEDs and the whole strip in altogether will have the same end state no matter if you have the strip.show() inside or outside the loop, pushing the data to the LEDs every cycle will result in an aesthetically much more pleasing animation . 

Arguing about what will be displayed on the strip ultimately is pointless and only takes us further away from my original question: Would it be possible to run the LEDs faster on RPi?

sschroe, 2017-11-14:
Running them faster is not possible, though the limitation comes from the LED protocol itself and not from this library or the RPi.

With 300 LEDs you should be able to get ~110Hz and with 900 LEDs ~37Hz.

penfold42, 2017-11-14:
One frame is 900x24 bits = 21600 bits
At about 800 kHz each bit takes 1.25 uSec
21600 x 1.25 = 27mS

Doing an led chase would take 24 seconds.
This is best case there’s overhead and it’s python

If you want it faster, you need to break your strip up.
PWM supports 2 channels = strips and you could use another instance via spi.

closed 2017-11-24 by jgarff