jgarff/rpi_ws281x #237: Golang binding issue (didn't work)

CriJonsi, 2017-11-10:
Hi ! I'm currently trying to control my leds (SK6812RGBW) with golang. 

First of all I installed all the package on my Raspberry pi 3. Then I ran the python code examples. Everything work fine!

Then I tried to run the golang (simple) code:
```
package main

import "ledRGBWTest/ws2811"

func main() {
	ws2811.Init(18, 3, 128)
	ws2811.SetLed(2, uint32(255))
}

```

The code didn't work I get this error:

```
fatal error: unexpected signal during runtime execution
[signal SIGFPE: floating-point exception code=0xfffffffa addr=0x0 pc=0x76ec62e8]

runtime stack:
runtime.throw(0xa70c7, 0x2a)
	/home/pi/go/src/runtime/panic.go:596 +0x70
runtime.sigpanic()
	/home/pi/go/src/runtime/signal_unix.go:274 +0x250

goroutine 1 [syscall, locked to thread]:
runtime.cgocall(0x84794, 0x10443f50, 0xf82c8)
	/home/pi/go/src/runtime/cgocall.go:131 +0xb4 fp=0x10443f38 sp=0x10443f1c
ledRGBWTest/ws2811._Cfunc_ws2811_init(0xf82c8, 0x0)
	ledRGBWTest/ws2811/_obj/_cgo_gotypes.go:132 +0x38 fp=0x10443f4c sp=0x10443f38
ledRGBWTest/ws2811.Init.func1(0xf82c8, 0x0)
	/home/pi/work/src/ledRGBWTest/ws2811/ws2811.go:51 +0x98 fp=0x10443f6c sp=0x10443f4c
ledRGBWTest/ws2811.Init(0x12, 0x3, 0x80, 0x104000f0, 0x104000f0)
	/home/pi/work/src/ledRGBWTest/ws2811/ws2811.go:51 +0x54 fp=0x10443fa4 sp=0x10443f6c
main.main()
	/home/pi/work/src/ledRGBWTest/main.go:6 +0x2c fp=0x10443fbc sp=0x10443fa4
runtime.main()
	/home/pi/go/src/runtime/proc.go:185 +0x1e4 fp=0x10443fe4 sp=0x10443fbc
runtime.goexit()
	/home/pi/go/src/runtime/asm_arm.s:1017 +0x4 fp=0x10443fe4 sp=0x10443fe4

goroutine 17 [syscall, locked to thread]:
runtime.goexit()
	/home/pi/go/src/runtime/asm_arm.s:1017 +0x4

```

The error come from this line:

`	res := int(C.ws2811_init(&C.ledstring))`

So my question are the following:
- Why the initialization didin't work?
- Why the python examples work and not the go binding?
- How can I fix this error?

Best regards!

CriJonsi, 2017-11-10:
Moreover I placed: 
- ws2811.h and pwm.h into /usr/local/include
- libws2811.a into /usr/local/lib

CriJonsi, 2017-12-18:
Of course I run the code with sudo

cpheps, 2017-12-28:
I'm getting the same issue @CriJonsi have you made any headway on it?

cpheps, 2018-01-03:
@jgarff Do you have any insight into this? I really would like to use this in golang. I'm running go 1.9 linux/arm and keep getting this issue. I was able to replicate exactly what @CriJonsi experienced with the same code.

cpheps, 2018-01-03:
@CriJonsi I got it working using your example with lite modifications. 

Here's the code I used
```

package main

import (
	"fmt"
	"ledRGBWTest/ws2811"
)

func main() {
	defer ws2811.Fini()
	err := ws2811.Init(18, 3, 32)
	if err != nil {
		fmt.Println(err)
	} else {
		ws2811.SetLed(2, uint32(0x000020))
                err = ws2811.Render()
                if err != nil {
                   fmt.Println("Render err " + err.Error())
               }
	}
}
```

I also had to run `go build main.go` then used `sudo` to run the resulting executable to get it to work.

CriJonsi, 2018-01-30:
I test your code but it didn't work (build and then sudo of course). I get this error:

`fatal error: unexpected signal during runtime execution
[signal SIGFPE: floating-point exception code=0xfffffffa addr=0x0 pc=0x76ea82e8]

runtime stack:
runtime.throw(0xa7998, 0x2a)
	/home/pi/go/src/runtime/panic.go:596 +0x70
runtime.sigpanic()
	/home/pi/go/src/runtime/signal_unix.go:274 +0x250

goroutine 1 [syscall, locked to thread]:
runtime.cgocall(0x84fb8, 0x10443f30, 0xf9298)
	/home/pi/go/src/runtime/cgocall.go:131 +0xb4 fp=0x10443f18 sp=0x10443efc
github.com/jgarff/rpi_ws281x/golang/ws2811._Cfunc_ws2811_init(0xf9298, 0x0)
	github.com/jgarff/rpi_ws281x/golang/ws2811/_obj/_cgo_gotypes.go:131 +0x38 fp=0x10443f2c sp=0x10443f18
github.com/jgarff/rpi_ws281x/golang/ws2811.Init.func1(0xf9298, 0x1c774)
	/home/pi/work/src/github.com/jgarff/rpi_ws281x/golang/ws2811/ws2811.go:51 +0x98 fp=0x10443f4c sp=0x10443f2c
github.com/jgarff/rpi_ws281x/golang/ws2811.Init(0x12, 0x3, 0x20, 0x0, 0x84ae4)
	/home/pi/work/src/github.com/jgarff/rpi_ws281x/golang/ws2811/ws2811.go:51 +0x54 fp=0x10443f84 sp=0x10443f4c
main.main()
	/home/pi/work/src/main.go:10 +0x48 fp=0x10443fbc sp=0x10443f84
runtime.main()
	/home/pi/go/src/runtime/proc.go:185 +0x1e4 fp=0x10443fe4 sp=0x10443fbc
runtime.goexit()
	/home/pi/go/src/runtime/asm_arm.s:1017 +0x4 fp=0x10443fe4 sp=0x10443fe4

goroutine 17 [syscall, locked to thread]:
runtime.goexit()
	/home/pi/go/src/runtime/asm_arm.s:1017 +0x4
`

For information I test with another library an I'm able to communicate with the led.

turino, 2018-02-28:
I'm having the exact same problem as @CriJonsi, also on pi 3. Would love to see a solution for this!

cpheps, 2018-03-01:
@CriJonsi @turino Did you try running the [Golang examples](https://github.com/jgarff/rpi_ws281x/tree/master/golang) I pushed awhile back to see if those are working? I don't remember doing anything special past what the README states in the Golang examples.

turino, 2018-03-01:
@cpheps yeah, I'm running this: https://github.com/jgarff/rpi_ws281x/blob/master/golang/examples/basic.go

and I get this:
```
fatal error: unexpected signal during runtime execution
[signal SIGFPE: floating-point exception code=0xfffffffa addr=0x0 pc=0x76f2ba6c]

runtime stack:
runtime.throw(0xb7b75, 0x2a)
	/usr/local/go/src/runtime/panic.go:605 +0x70
runtime.sigpanic()
	/usr/local/go/src/runtime/signal_unix.go:351 +0x24c

goroutine 1 [syscall, locked to thread]:
runtime.cgocall(0x90cfc, 0x1134def0, 0x119d48)
	/usr/local/go/src/runtime/cgocall.go:132 +0xb8 fp=0x1134ded8 sp=0x1134debc pc=0x13424
github.com/jgarff/rpi_ws281x/golang/ws2811._Cfunc_ws2811_init(0x119d48, 0x0)
	github.com/jgarff/rpi_ws281x/golang/ws2811/_obj/_cgo_gotypes.go:132 +0x38 fp=0x1134deec sp=0x1134ded8 pc=0x8fba4
github.com/jgarff/rpi_ws281x/golang/ws2811.Init.func1(0x119d48, 0x1134def8)
	/usr/local/go/src/github.com/jgarff/rpi_ws281x/golang/ws2811/ws2811.go:51 +0x84 fp=0x1134df08 sp=0x1134deec pc=0x90124
github.com/jgarff/rpi_ws281x/golang/ws2811.Init(0x12, 0x10, 0xff, 0x1dad8, 0x11352070)
	/usr/local/go/src/github.com/jgarff/rpi_ws281x/golang/ws2811/ws2811.go:51 +0x54 fp=0x1134df40 sp=0x1134df08 pc=0x8fda0
main.main()
	/home/pi/rpi_ws281x/golang/examples/basic.go:19 +0x48 fp=0x1134dfb8 sp=0x1134df40 pc=0x907d0
runtime.main()
	/usr/local/go/src/runtime/proc.go:195 +0x21c fp=0x1134dfe4 sp=0x1134dfb8 pc=0x3a5cc
runtime.goexit()
	/usr/local/go/src/runtime/asm_arm.s:971 +0x4 fp=0x1134dfe4 sp=0x1134dfe4 pc=0x60150
```

cpheps, 2018-03-01:
I'll check and see later if I did anything that's not documented.

turino, 2018-03-08:
@cpheps any update? Thanks in advance!

cpheps, 2018-03-09:
Sorry for the late response. I can't seem to find anything that I did different than what I posted in the README for the golang. I'll do some further investigation see if I can't turn anything up but as of now sorry I don't have any insight.