jgarff/rpi_ws281x #238: Jupyter Notebook Runtime Error

JAICHANGPARK, 2017-11-17:
Hello?

Now, I'm using Jupyter notebook in RSP 3 on python3

I having error this code line
  strip.begin()
```
RuntimeError                              Traceback (most recent call last)
<ipython-input-24-e0494937b86a> in <module>()
      4         strip = Adafruit_NeoPixel(LED_COUNT, LED_PIN, LED_FREQ_HZ, LED_DMA, LED_INVERT, LED_BRIGHTNESS, LED_CHANNEL, LED_STRIP)
      5         # Intialize the library (must be called once before other functions).
----> 6         strip.begin()
      7 
      8         print ('Press Ctrl-C to quit.')

/usr/local/lib/python3.5/dist-packages/rpi_ws281x-1.0.0-py3.5-linux-armv7l.egg/neopixel.py in begin(self)
    113                 if resp != ws.WS2811_SUCCESS:
    114                         message = ws.ws2811_get_return_t_str(resp)
--> 115                         raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, message))
    116 
    117         def show(self):

RuntimeError: ws2811_init failed with code -5 (mmap() failed)
```
How to fix this problem?

penfold42, 2017-11-17:
It probably needs to run as root

JAICHANGPARK, 2017-11-17:
@penfold42 
Thank you , I was run jupyter notebook --allow-root 
But, I had Runtime error.. Is it correct?

Gadgetoid, 2021-12-03:
Closing since this is stale and I like the number of issues to go down sometimes :laughing: instead of just up :sob: 

Feel free to re-open if it's still relevant.

closed 2021-12-03 by Gadgetoid