jgarff/rpi_ws281x #240: RPi3 not seen as raspberry pi

mikaye, 2017-11-17:
I get the following when running on RPi3,
[rpi-ws281x-native] it looks like you are not running on a raspberry pi so there will be no functionality exposed by this module. For convenience, implementation stubs are provided.

What can I change in board_info.c in order to support this pi and by updating board_info.c to return true, can I expect this to work? Here's the RPi3's cpuinfo,

processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a02082
Serial		: 0000000001e5039e

penfold42, 2017-11-17:
Where do you install from ?

I can’t find any mention of that error.
The boards are defined in rpihw.c
Yours is included - hwver  = 0xa02082,

mikaye, 2017-11-18:
Sorry, not your issue. This came from https://github.com/raspberry-node/node-rpi-ws281x-native. Your work was mentioned here and I mistakenly assumed it was using it directly.

penfold42, 2017-11-18:
It appears to be using a 2 year old fork of this

jgarff, 2017-11-24:
Closing issue.  The downstream repo will need to pull latest version.

closed 2017-11-24 by jgarff