jgarff/rpi_ws281x #243: Issue When Powering Through Pins 2 and 4

xShojirox, 2017-11-25:
So I am working on using a Raspberry Pi for a controller for a strand of WS2811 lights for a Christmas tree. I am trying to power the Pi through the 5v pin 2 and/or pin 4 so that I don't have to have an additional power cable going out and using an additional outlet. I plan to use a Raspberry Pi Zero W once I have my current arduino code converted over to python, but in the meantime I am using my Pi 3, as it already has the pins on it.

The issue that I am finding though is that everything works perfectly when I am powering the Pi 3 from micro usb, but when I power it from pin 2 and/or 4 there are issues. It powers on, and I can connect to it, but when I try to run the strandtest the lights seem to flicker randomly in random colors. It happens even if the pi is getting power from the micro usb port as well. After removing power from pin 2 and 4 and running the strandtest it works just fine.

I haven't tested it on the zero w, so I am not sure if this issue will affect it as well, but there is a chance that I end up just going with the pi 3 for other reasons, and I would like to find out if there is a way around the issue. I also haven't tried, but I assume it might work if I cut up a micro usb cable and connect my 5v power source through that.

Before anyone asks, yes I have disabled audio, and I am using a level shifter. It works just fine when powered through micro usb and no power is going into pins 2 and 4.

I understand the issue itself that I am having may be something that isn't resolvable, I just wanted to see if anyone knew of any possible way to get around it.

Gadgetoid, 2018-03-06:
I believe your issues are wiring/electrical, rather than something awry with this library. Are you still having problems?

When powering the Pi through pins 2 and/or 4, are you connecting ground?

penfold42, 2018-03-08:
How stable, and noisy is the 5v on the gpio pins ?

xShojirox, 2018-03-09:
Unfortunately I don't have the equipment right now to test that, I should be able to test it within the next few weeks though. That said, it is the same power source I use to power an arduino based board along with the same set of lights (not at the same time as testing this of course). so I feel like it can't be too unstable or noisy, otherwise that would be having issues.

gilvad, 2018-06-25:
on Raspberry Pi Zero W the pin 2 and 4 are connected to USB power in.
On Raspberry Pi 3 the power comes from a regulator and it might be slightly higher.

Data needs to be at least 0.7 of the 5 Volt which means that normal 3.3Volt output should not work with 5.0 and above. You must use a 74HC type buffer (and gate or other) to output a proper data signal from Raspberry. Alternatively you can use a diode in series with the 5V for the first LED which will drop it to 4.4-4.5 Volt where the Raspberry GENIO output will be good enough. The LED controller chip will then send the data to the next in the chain with higher data value. The following LEDs can then use +5V and get full brightness.

I tested this today with 3 meter wires and it works fine.