jgarff/rpi_ws281x #244: Ws2812 LEDs do only show 1 Color

SillyScientists, 2017-11-29:
After I have installed the libary and started to run the strandtest.py. I just got that issue that everything that is played is in red. So there is no other color than the red ones.
I hope you can help me.
Thanks in advance

Dominik Schumann

SillyScientists, 2017-11-29:
Now flickers every LED that isn't on.

sschroe, 2017-11-29:
Please provide more information on the hardware setup and the solutions you already tried from the README (audio blacklisting, ...)

SillyScientists, 2017-11-30:
Using Rpi 2B 
With 240 LEDs of WS2812B Used the strandtest.py with editing the LED Count at First.
Changed after then the Frequency to 400kHz.
Nothing changed.
Edited the Channel as Well changed nothing

Before I Used the libary I blacklisted the audio and as the HDMI thingy isn't for the Rpi 2B, I Just ignored it.

Should I use the SPI Channel ? 
Which PIN should I use and should I Change any settings ? 

Thanksgiving in advance

SillyScientists, 2017-11-30:
So I just used the SPI PIN and the structures etc. works fine, but the color will constantly be on red.

sschroe, 2017-12-02:
What about the used voltages and a level shifter?

SillyScientists, 2017-12-02:
I use 5 V ans 10 A 
A Level shifter ? 
I don't have anyone

sschroe, 2017-12-03:
In that case grab something like an 74HCT125 and add it.

SteMMo33, 2019-12-11:
I compiled the library and the test app for Tinker, forcing SPI port.
The application is running but the signal is regular at first (let's say 8 LEDs) then the signal is larger and irregular.
Of course the LEDs are blinking...

Gadgetoid, 2021-12-03:
Closing since this is stale and I like the number of issues to go down sometimes :laughing: instead of just up :sob: 

Feel free to open a new issue the Python bindings if it's still relevant- https://github.com/rpi-ws281x/rpi-ws281x-python

closed 2021-12-03 by Gadgetoid