jgarff/rpi_ws281x #245: How to maintain the other LED lights on after addressing a specific one

Icsaac, 2017-12-07:
Hi. Im doing a project, in which i am addressing a individual LED light at the time, but when I run again the program addressing to another LED light, the one I previously addressed, turns off. 

Is there a way to maintain other LED lights on, by addressing one individually without the other ones turning off ?? :/

penfold42, 2017-12-07:
No

Icsaac, 2017-12-07:
Is there any way to do this ??

penfold42, 2017-12-07:
You need to keep track of the current state yourself.

If you need to do this over multiple invocations I suggest you write it to a file to hold the last/initial state

Icsaac, 2017-12-07:
Yeah, i was thinking about doing that with the getPixelColor() function.

Gadgetoid, 2018-03-06:
@Icsaac `getPixelColor` is incapable of reading the state back from the LEDs, it will only read state back from the memory buffer in the currently running instance of the library. If your script exits, this state is lost.

The LEDs are non-addressable, and write-only. You *must* keep track of the pixel states yourself separately as @penfold42 suggests.

closed 2018-03-06 by Gadgetoid