jgarff/rpi_ws281x #246: NeoPixel rung not working properly with RPi Zero W

russellseymour, 2017-12-07:
I have connected a RPi Zero " to a 16 pixel NexPixel ring. I am using SPI to control the LEDs so the data line is connected to pin 10 on the Pi0W.

I have successfully compiled the library and am trying to use the python library with the `examples/strandtest.py` script.

The issue is that at the beginning things looks to be working, e.g. a colour wipe goes around the ring. However shortly after this, all the LEDs display white and nothing changes. I do not see the rest of the demos in the script.

These are the settings I have in the script

```python
LED_COUNT      = 16      # Number of LED pixels.
#LED_PIN        = 18      # GPIO pin connected to the pixels (18 uses PWM!).
LED_PIN        = 10      # GPIO pin connected to the pixels (10 uses SPI /dev/spidev0.0).
LED_FREQ_HZ    = 800000  # LED signal frequency in hertz (usually 800khz)
LED_DMA        = 10       # DMA channel to use for generating signal (try 5)
LED_BRIGHTNESS = 64     # Set to 0 for darkest and 255 for brightest
LED_INVERT     = False   # True to invert the signal (when using NPN transistor level shift)
LED_CHANNEL    = 0       # set to '1' for GPIOs 13, 19, 41, 45 or 53
LED_STRIP      = ws.WS2811_STRIP_RGB   # Strip type and colour ordering
```

I have tried changing the DMA from 5 to 10 etc but that has not made any difference.

Have I got the LED_STRIP wrong or something? It is almost there, just not sure why it is not working. ANy help gratefully recieved.

sschroe, 2017-12-07:
What Voltage do the LEDs run with?
Is a level shifter present?
Did you try the audio blacklisting from the readme?

russellseymour, 2017-12-08:
I apologise for raising this issue, it is completely user error.

@sschroe Your suggestion of blacklisting the audio as in the README made me read it again, and I realised that I had forgotten to add some settings to the `/boot/config.txt` when using SPI.

I have now made these changes and it is working now.

(Note to self - RTFM)

closed 2017-12-08 by russellseymour