jgarff/rpi_ws281x #247: How to get the RGB value of a LED using getPixelColor() URGET :(

Icsaac, 2017-12-07:
Hi, 

Ive been trying to get the RGB value of a specific LED light using the function getPixelColor().But when I try to print the function value, the only thing a I get is 0. Any help with this. Im using this simple code:

 for i in range(strip.numPixels()):
                value=strip.getPixelColor(i)
                print(value)

penfold42, 2017-12-07:
Did you set any colours beforehand ?

Icsaac, 2017-12-07:
Yeah, all my strip to white (10 LEDS) RGB(255,255,255)

Gadgetoid, 2018-03-06:
Could you post the code you're using to set the pixel colours?

I've run quick test with the latest library code, and I can't seem to replicate this.

Gadgetoid, 2018-03-06:
I just saw #245 - if you are expecting `getPixelColor` to read the colour of an LED back from the hardware, you'll be disappointed. The LEDs are write-only. `getPixelColor` can only read back the state from the libraries memory buffer.

You *must* either save your state externally, or set your script up to run continuously.

Note: It's not generally a good idea to write pixel state and expect the hardware to maintain the colour you have set. It's easy for electrical interference to be interpreted by the LEDs as signal, and this may result in random, unexpected changes. You should continuously refresh the pixels with the desired state.

Gadgetoid, 2018-08-28:
Since there has been no input on this for a few months, I'm closing it for housekeeping. Feel free to re-open if you have further questions or comments.

closed 2018-08-28 by Gadgetoid