jgarff/rpi_ws281x #252: Make Python code follow PEP8 coding standards.

RichardBronosky, 2017-12-26:
[PEP8](https://www.python.org/dev/peps/pep-0008/) is the "Style Guide for Python Code" accepted by most teams/projects and implemented by most linters.

I discovered that in https://github.com/jgarff/rpi_ws281x/blob/master/python/examples/strandtest.py after line 37 indention switches from spaces to tabs. That is generally pretty disgusting, but even worse in Python code. I saw from https://github.com/jgarff/rpi_ws281x/issues/23 that the function naming shouldn't be changed, so I won't bother with that.

If I were to go through the trouble to modify all the Python with what is likely to be mostly whitespace changes, would it be accepted? I don't want to bother otherwise.

Gadgetoid, 2018-03-06:
I'm hoping to render the Python code here obsolete in favour of these bindings: https://github.com/pimoroni/rpi_ws281x-python.

I don't expect you to do so. But if you were to make the suggested changes to these examples and migrate them to an examples folder in `rpi_ws281x-python` I would graciously accept your efforts.

For more details upon why I want to accomplish this, see https://github.com/jgarff/rpi_ws281x/issues/267

And FWIW I'm sure your efforts would be appreciated here, too. I'm a very strong believer in embracing PRs for typos and general code formatting.

Gadgetoid, 2018-08-29:
Finally implemented in: https://github.com/rpi-ws281x/rpi-ws281x-python/pull/11

closed 2018-08-29 by Gadgetoid