jgarff/rpi_ws281x #260: Running without root/sudo

ffleandro, 2018-01-04:
Is it possible to run the `./test` without sudo?

I've seen several issues mentioning this problem, but none addressing it directly.
I can successfully run the C code *WITH* sudo.
But in my application I need to use Python and can't run it with sudo.

What exactly is the limitation that makes `sudo` mandatory and is there any workaround without raising any security issues?

In this forum [thread](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=22515#p216411) there is an explanation on why changing permissions of `/dev/mem` won't work, but I'm a bit out of my league here.

Any help?

ffleandro, 2018-01-05:
This issue is being addressed in #251

ffleandro, 2018-01-05:
I have successfully used this library through SPI by changing the access in `mailbox.c` from `/dev/mem` to `/dev/gpiomem`.

### *WARNING:* This will make the PWM communication method unusable!

PandorasFox, 2018-01-05:
I fixed that in #257, which'll be merged shortly.

Gadgetoid, 2018-08-28:
I think this is fixed? Closing, but happy to re-open if any further issues arise.

closed 2018-08-28 by Gadgetoid