jgarff/rpi_ws281x #263: running on python 3.5 provides 'ws2811_t * no destructor found'

HH0718, 2018-01-18:
```
Traceback (most recent call last):
  File "main.py", line 22, in <module>
    LED_STRIP)
  File "/usr/local/lib/python3.5/dist-packages/rpi_ws281x-1.0.0-py3.5-linux-armv6l.egg/neopixel.py", line 79, in __init__
    ws.ws2811_channel_t_brightness_set(self._channel, brightness)
TypeError: in method 'ws2811_channel_t_brightness_set', argument 2 of type 'uint8_t'
swig/python detected a memory leak of type 'ws2811_t *', no destructor found.

```
I can run it if I use python 2.x but not 3.5. I have re-installed and rebuilt globally but I continue to get this error.

HH0718, 2018-01-18:
Turns out I had a variable that I set to divide and it would crash. Fixed bu using whole numbers instead.

closed 2018-01-18 by HH0718

glenneroo, 2019-04-02:
Are you sure that was the culprit? I'm having the same issue but I get this error upon exiting.

nicofirst1, 2020-07-02:
I'm getting the same error when trying to run a custom script where I initiate the class as:
```
from rpi_ws281x import PixelStrip
np= PixelStrip(pixels,PIN) 
np.begin()
```
I am working on Raspberry Pi 4 Model B Rev 1.2 with  Python 3.7.3 and running  the script with sudo.
The error is the following:
```
Traceback (most recent call last):
  File "src/raspberrypi/main.py", line 23, in <module>
    app = pat(handler=PiHandler, rate=rate, pixels=64, color=color)
  File "/home/pi/Repos/ledypie/src/patterns/chasing.py", line 13, in __init__
    super().__init__(**kwargs)
  File "/home/pi/Repos/ledypie/src/patterns/default.py", line 19, in __init__
    self.handler=handler(rate)
  File "/home/pi/Repos/ledypie/src/raspberrypi/pi_handler.py", line 11, in __init__
    self.np= PixelStrip(pixels,PIN)
  File "/usr/local/lib/python3.7/dist-packages/rpi_ws281x/rpi_ws281x.py", line 95, in __init__
    ws.ws2811_channel_t_count_set(self._channel, num)
TypeError: in method 'ws2811_channel_t_count_set', argument 2 of type 'int'
swig/python detected a memory leak of type 'ws2811_t *', no destructor found.
```

danpe, 2022-08-05:
For me it happens straight from the example  https://github.com/rpi-ws281x/rpi-ws281x-python/blob/3f7f031b8a87f584fcef72225b7cfb1cff682316/examples/lowlevel.py#L65