jgarff/rpi_ws281x #269: mmap() error: Operation not permitted -- Alpine Linux

JMurph2015, 2018-02-04:
So in an attempt to strip out everything unnecessary from my application (it's a networked LED controller), I tried getting it to work on Alpine Linux running on my Raspberry Pi 3.  I already have set up a new cross compiler on my laptop (arm-linux-musleabihf) and rebuilt my application with that compiler, but now the problem is that even when running as _root_, I get this initialization error saying "mmap() error: 
```
ledrpi1:~# ./rusty_controller 
Starting rusty_controller...
mmap error
: Operation not permitted
ws2811_init failed: 0x97e60

thread 'main' panicked at 'Init failed', src/main.rs:80:3
note: Run with `RUST_BACKTRACE=1` for a backtrace.
ledrpi1:~#
```
There's some Rust boilerplate at the end there that can be safely ignored, I'm worried about the lines right after "Starting rusty_controller".  If anyone has an idea of why this might be happening, be sure to let me know, because otherwise this Alpine is great for this, there's all of like 10 kernel modules loaded and it boots up really quick.

Gadgetoid, 2018-08-28:
I trust you got this up and running? I'll close this issue for now, but if you continue to have problems feel free to re-open.

closed 2018-08-28 by Gadgetoid

JMurph2015, 2018-08-28:
Yeah I got this mostly up and running, but I think I may have had to use some workaround.