jgarff/rpi_ws281x #270: Cant get strandtest.py to run without root

Quad14, 2018-02-07:
Hello,
I am trying to run strandtest.py without root access as a test for a latter script. I have enable spi on the rpi 3 and set the core_freq=250. I also changed LED_PIN = 10. 
I receive the following error:
python strandtest.py
Can't open /dev/mem: Permission denied
Traceback (most recent call last):
  File "strandtest.py", line 102, in <module>
    strip.begin()
  File "build/bdist.linux-armv7l/egg/neopixel.py", line 106, in begin
RuntimeError: ws2811_init failed with code -13 (Unable to initialize SPI)

Is there something I need to change in strandtest.py?
Thanks!

ErikBorra, 2018-02-09:
See https://github.com/jgarff/rpi_ws281x/pull/257

Quad14, 2018-02-12:
I followed (to the best of my understanding) those instructions but was still not able to get the strandtest.py to work on spi without sudo access. I should be able to use the neopixel.py ported library correct?

Gadgetoid, 2018-03-06:
Your error appears to be with enabling SPI, rather than the normal issue with root access required for `/dev/mem`.

Are you still having issues?

What do you see if you: `lsmod | grep spi`?

melzoghbi, 2018-06-08:
I am having the same issue on the latest Raspbian as well. i want to be able to run this without sudo. 
Yes i am a member of gpio group as well.

In my case:
sudo python strandtest.py    --> WORKS

python strandtest.py  --> THIS DOESN'T WORK

Here is the error i get:
RuntimeError: ws2811_init failed with code -5 (mmap() failed)

@Gadgetoid : This is the output when i run lsmod command:

[image]

Thoughts?

Gadgetoid, 2018-08-28:
@melzoghbi unless you've modified `standtest.py` to use an SPI-enabled pin, then the `mmap()` error is an expected outcome. You *must* run with `sudo` to supply privileged hardware access.

Gadgetoid, 2018-08-29:
Have duplicated this for posterity over on the Python bindings: https://github.com/rpi-ws281x/rpi-ws281x-python/issues/9

For all future issues/requests relating to Python please see: https://github.com/rpi-ws281x/rpi-ws281x-python/issues/

closed 2018-08-29 by Gadgetoid