jgarff/rpi_ws281x #272: Flickering in end strands, low voltage control signal

shaleaman, 2018-02-24:
I am using some WS2811 cabled leds to create an LED matrix. 

My wiring diagram is here:
[image]

and a video of the flicker:
https://imgur.com/1rMUtbJ

The framerate doesn't quite do it justice, but they all just tweak out for a millisec.

I have 3 panels, 4 strips of 12 per panel. The strips were cut to 12 LED's long and spliced. There is power injection for each panel. Power is fine, 5V all the way through. What I'm noticing is that there is a lot of flicker in the last panel and a little flicker in the second panel. Doesn't matter the order of the panels, the last one is always the flickery-est. 

I've tried to track down any bad solder, loose connections, bad grounding but everything works fine. The exact same scripts (the example script "strandtest.py" causes flicker) worked before I cut and spliced the cables, though I wasn't power-injecting during the tests before splicing so the last leds were very dim.

So, I considered that the control signal might be too low. I checked the voltage with a Multimeter and at the breadboard the signal after the level shifter was about 1V. After the first board it was around 0.5-0.6V. The second 0.4V, and at the end of the third it was at 0.3V. 

The voltage from the Pin seems to change based on the animation, but I am concerned the splicing may have caused too much of a voltage drop. I am using pretty decent wires for the power, but am using jumper wires, like you use on a breadboard, to splice the data signal wires. 

Am I on the right track thinking of voltage drop, or is there a more likely reason?

Thanks

SlySven, 2018-02-24:
No, my guess, is that because each LED removes a chunk of data from the control line as it passes between D<sub>in</sub> and D<sub>out</sub> and on to the next device and the proportion of the time that the line is HIGH falls off.  Either you need to set your (Digitail) VMM to MAX mode - or you need to hook up to an oscilloscope to actually see the waveform...

shaleaman, 2018-02-24:
Hmm, what is VMM? Google fails me there.

mbelling, 2018-02-24:
What version of Raspberry Pi are you using?  I noticed a flicker similar to this with my lights and a Raspberry Pi 3.  The problem ended up being that the application was using DMA 5 (The default in the python strandtest), which seems to cause problems on a Pi 3.  I switched to 10 and the flicker went away.  It's not guaranteed to be the same problem, but might help track down yours.

shaleaman, 2018-02-26:
@mbelling Yeah, I've changed DMA to 10 and the flicker still happens. 

I have, though, learned how to make the flickering much worse. By pushing the power to all the LEDs up to 90 to 100% white, the last two panels have near constant flickering, with the last panel perhaps flickering a little more than the second. ALSO, if I grab and wiggle the data wire in my breadboard, the exact flicker pattern (panel 1 ok, panel 2 some flicker, panel 3 lots of flicker) intensifies. I probably need to solder all of this onto a prototyping board to secure the connections rather than using jumpers and a breadboard, but, being a burgeoning electronics hobbyist, I was wondering what exactly is causing the flicker. It seems to be a connection issue, but what is happening to the signal, especially at higher intensities, that is causing the flicker?

penfold42, 2018-02-26:
Do you have any 74HCTxx chips lying around ?

I’ve had issues with mosfet bidirectional level shifters. 

It could also be a power issues - a couple of 47-100uF caps along the strips might help

shaleaman, 2018-02-26:
Ok, I realize this is beyond the scope of the software at this point, but I am a little desperate and I've found new things about the flicker.

- removing panel 3 drastically reduced the flicker in panel 2 when the LED brightness is closer to 255 (flicker still only happens when data is flowing). 

- touching the data wire from the breadboard to the first LED causes a full-string color flicker freakout. The wire is sheathed and it only freaks out with skin contact. Touching it with a pen or anything else does not produce the effect. Touching any other wire doesn't do it either, but touching the level-stepper output pin causes severe dimming.

@penfold42 I'm bringing in some 74AHCT125  tomorrow, hopefully that is the issue.

Thanks!

eloots, 2018-02-26:
On my set-up (using a single 8-LED NeoPixel strip), I also had problems with LED intermittently flickering.

The solution I applied was to reduce the 5V power supply to the strip by putting a simple (10 cent) diode (1N4001) will do between the 5V power supply and the 5V input connection on the strip. What this does is that it reduces the voltage fed to the strip by 0.7V (so bringing it to 4.3V). Apparently, this brings the 3.3V data signal in margin. Had no flickering since I applied this on 25 boards each equipped with one LED strip.

shaleaman, 2018-02-27:
I've seen setups with the diode online as well. Thanks for reminding me about that.

Ok, got a 74AHCT125 Level Shifter and a 1N4001 diode coming in tomorrow. Somethings gotta work.

shaleaman, 2018-02-28:
So it seems the connection between the panels was the key. I got the new level shifter and that did not help, and the diode sorta got super hot and melted through the terminal block plastic. Not sure what I did wrong there, but I'm putting that aside now.

I had a hunch about resistances (when I held the two connections between the panels with my fingers, the flicker went away) and put 330ohm resistors between the data signal of each panel. That seems to have done it. The flicker is gone even at the highest brightness. 

I've put 47 to 1K resistor between the level shifter and the D<sub>out</sub> and the flicker was fairly unaffected.

So I'm glad it works, but I don't know why and I'm guessing this is not the best way to handle this. Does anyone have any suggestions for cleaning this up?

Thanks!

eloots, 2018-02-28:
> I've put 47 to 1K resistor between the level shifter and the Dout and the flicker was fairly unaffected.

Cool, the idea about putting a small resistor between strips! That’s between Dout from the previous strip and Din of the next one, right?
I guess that this reduces the appearance of signal glitches on Din which would trigger the spurious flashing.

The power-loss in the diode can be an issue: if all strips are fed through a single diode and you have many strips at full brightness, the power dissipation in the diode can be very high. Say you have 1 Amp total current, you have 0.7 Watt (1A x 0.7V). In this way you may exceed the max. current/dissipation specs of the diode.

Simple alternative is to utilize one diode per strip. 

Verstuurd vanaf mijn iPhone

penfold42, 2018-02-28:
Ouch - 144 leds x 60mA

9 amps thru a 1 amp diode

eloots, 2018-02-28:
That poor diode must have been red-hot. It’s probably dead now...

SlySven, 2018-02-28:
Or lost any ability to stop the flow of electrons in one direction - on the plus(?) side the current handling capacity has probably increased significantly now although there will no longer be much of a voltage drop across the short-circuit...

⚰️ ... it is a dead diode, it has gone to meet it's maker, it has, for all intents and purposes, shuffled off this mortal coil.

shaleaman, 2018-02-28:
Ha, yeah, fried diode and no fire is a cheap lesson :) Luckily I bought 100 for 3 bucks. 

So I'm not sure how the power along the strips interacts with the D line, but it seems that power injecting is causing the data line to get scrambled, since the flickering only happens after the power injection. I'm guessing wherever a power injection happens, a resistor must be added to the D<sub>IN</sub>

According to the [spec sheet](https://cdn-shop.adafruit.com/datasheets/WS2811.pdf): To prevent the reflection and hot-swap protection, we suggest to connect a 33ohm resistor at the data input or output port for impedance

closed 2018-02-28 by shaleaman