jgarff/rpi_ws281x #274: SK6805_2427 LED Do not work

PatrikWey, 2018-03-05:
Hi
I’m using successfully a “unicorn pimoroni unicorn phat” and the corresponding node Red Integration. 
Now because the shape is not pleasant, for final design I made my own PCB.
But instead of the WS2812 (neopixels) on the Hat I’m using the SK6805-2427 because of the smaller size, They have the same protocol and timing like the SK6812 LED and are sold as neopixels like the WS2812 and SK6812 LEDs.

I made 10 boards and 5 of them seems to work fine. But 5 of them are showing random Colors.
I made some measurements and noticed the control signal from the pi is not according to the Datasheet from WS2812, SK6812 or SK6805. As a future point I noticed the datasheets are different but both are sold as neopines.
Is there a way to mod the control Signal? (mainly low high timing, frequency is fine) To make my SW work with the SK6805 LEDs and the same Lib?
(I’m using the node-red integration and do not like to establish new interfaces)
Thanks for Feedback
Patrik

penfold42, 2018-03-05:
What level shifter are you using ?

PatrikWey, 2018-03-05:
Hi
I’m using MC74VHC1GT08DTT1G (level shifting and gate) in my circuit, for an experiment I used 74LVC1G17W5-7 (level shifting buffer). It makes no difference.
And If I measure this by scope the Signal from the pi is not disturbed delayed or something else only level shifted from 3.3 to 5V. This should be ok. 
I do not believe it can be from the level shifter.

I also made an experiment. I added a WS2812 LED as pixel 0 to my circuit and added my SK6805 LED circuit after this WS2812 LED. This makes my circuit Working. The WS2812 LED works now as a timing correction. The WS2812 is shifting the timing close to his datasheet spec and because the values are (more or less) in the middle of the spec, it is now compatible to the SK6805 LED and OK.
It really seams the SK6805 LED are not working with the timing created by the Pi.
I studied the code and the datasheet. I suggest (without going deeper) the code creating a High by Signal array 110 and a Low by 100, this need to be driven by 2,4 Mhz.
But to fulfill the Datasheet spec we would need to create this 111100 for High and 111000 for Low and this would need to be driven by 4.8 Mhz.

What do you think?

PatrikWey, 2018-03-05:
Sorry correction first time I was to fast 😊
I studied the code and the datasheet. I suggest (without going deeper) the code creating a High by Signal array 100 and a Low by 110, this need to be driven by 2,4 Mhz.
But to fulfill the Datasheet spec we would need to create this 110000 for High and 111000 for Low and this would need to be driven by 4.8 Mhz.

penfold42, 2018-03-05:
I would start by changing the default frequency. This may be sufficient to alter the timing to suit.

Use the test app that gets built - main.c uses the default 800000. Try +/- 50,000 or 100,000

PatrikWey, 2018-03-05:
Hi
Thanks for feedback.
OK I’m going to try. 

What is the easiest way to do this? 
I have installed the Lib by using “curl -sS get.pimoroni.com/unicornhat | bash”
And the node Red node via npm “npm i node-red-node-pi-unicorn-hat”

But I’m a bit afraid about the Temperature effect if I do it like this. I’m using the device from -5 to around 30 °C. May it would be good to solve this issue in a proper way. If this is possible.
Thanks for feedback

PatrikWey, 2018-03-05:
I made your recommended frequency test.
If I go to 750 or 700kHz it gets even worse.

If I go to 825kHz it seams to work. 😊
But the signals and frequency is not as expected the frequency changes to 914kHz (if set to 800kHz)!

Enyway I’ll try this wor around for now.

Has Enyone an idee how to combine this with the node red node? Or knows how I can implement this bugfix into the Node red node?

Thanks for feedback

penfold42, 2018-03-05:
I’m not sure frequency measurements mean a whole lot.

It’s the high time for the zero and one bits that’s most critical.

You can change the default frequency in ws2811.h but the library should also take the frequency at runtime

Gadgetoid, 2018-03-06:
> Has Enyone an idee how to combine this with the node red node? Or knows how I can implement this bugfix into the Node red node?

You'd need to modify the Unicorn HAT python library- if you're using the Node RED node we ship - and edit the LED_FREQ_HZ value passed in during initialisation:

https://github.com/pimoroni/unicorn-hat/blob/84cdd47684a8ca8102fa0ed94ff1539abf37c01d/library/UnicornHat/unicornhat.py#L10

You can clone the Unicorn HAT GitHub repository, edit this file, and use `sudo python setup.py install` to replace your local Unicorn HAT library. (Substituting `python3` if appropriate)

PatrikWey, 2018-03-06:
Hi 
I tried following your advice.
I did the following:
Node Red
•	Open node red and open Palete management and uninstall unicorn hat node
•	Node-red-stop

WS281x lib
•	git clone https://github.com/jgarff/rpi_ws281x.git
•	cd rpi_ws281x
•	nano ws28.11.h // and set the frequenzy from 800000 to 825000
•	sudo scons
•	sudo ./test 
this maks my pixel working right! @914kHz
Stop script

unicornhat
•	git clone https://github.com/pimoroni/unicorn-hat
•	cd unicorn-hat/library/UnicornHat
•	nano unicornhat.py  // and set the LED_Freq_Hz from 800000 to 825000
•	sudo python setup.py install
•	cd
•	cd unicorn-hat/examples
•	sudo python demo.py
this does not work and switches my pixel back to 800kHz !

Please some advice.

Gadgetoid, 2018-03-07:
You might need to modify the underlying version of rpi_ws281x used by Unicorn HAT, then.

Something like:

```
git clone https://github.com/pimoroni/rpi_ws281x-python
cd rpi_ws281x-python
git submodule update --init
cd library/lib
nano ws2811.h // and change frequency
cd ../
sudo python setup.py install
```

PatrikWey, 2018-03-07:
Hi
I finaly made it working. 😊

I did not notice the Script 
curl -sS get.pimoroni.com/unicornhat | bash
is writing in different folders and is installing the WS281x lib by its own.

It is not necessary to make a git clone and reinstall (it makes it even worse because you wil have 2 independet librarys after this)

The script creates 3 directoris:
•	Pi/pimoroni/unicorn-hat/…
•	/user/local/lib/python_2.7/dist…..
•	/user/local/lib/python_3.5/dist…..
In the second 2 folders you will find 2 files where the frequency is specified.
Ws281x.py and unicron.py

If you edit the 800000 to 825000  and reboot the pi
The example scrips are working with 914kHz

now you can reinstal the node red node.

Thanks for advice!

Gadgetoid, 2018-03-07:
No problem! Glad you're up and running.

closed 2018-03-07 by Gadgetoid