jgarff/rpi_ws281x #275: Is it not possible to install Neopixel library on Pine64 with aarch64

ulpin, 2018-03-10:
Im getting the following error while trying to install on Pine64 with aarch64 architecture (OS: ubuntu mate). Am I doing something wrong or isn't it suppose to work?

`ubuntu@pine64:~/rpi_ws281x$ nano SCons
SConscript  SConstruct  
ubuntu@pine64:~/rpi_ws281x$ nano SConscript 
ubuntu@pine64:~/rpi_ws281x$ scons
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
CC      mailbox.o
mailbox.c: In function 'unmapmem':
mailbox.c:71:25: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
     uint32_t baseaddr = (uint32_t)addr & pagemask;
                         ^
mailbox.c:74:16: error: cast to pointer from integer of different size [-Werror=int-to-pointer-cast]
     s = munmap((void *)baseaddr, size);
                ^
cc1: all warnings being treated as errors
scons: *** [mailbox.o] Error 1
scons: building terminated because of errors.
`

penfold42, 2018-03-10:
It won’t work.

this directly hits the hardware and it’s different on a rockchip SoC.

ulpin, 2018-03-10:
Ok.
Seems I'm screwed to get my WS1812 LEDs running on the Pine64.

And thanks for the fast response.

closed 2018-03-10 by jgarff

penfold42, 2018-03-10:
Other options:
1) If the kernel supports SPI _with_ DMA it’s possible to hack this library to remove some of the lingering hardware setup code.
A stand-alone test I wrote is at: https://github.com/penfold42/stuff/blob/master/ws2812_spi.c

2) use a cheap arduino and hang them off USB

3) work out how to drive the PWM hardware on the Rockchip SoC