jgarff/rpi_ws281x #277: One LED always on

umbertix, 2018-03-16:
Tl;DR
What appears to be the first led is always on, I can't turn it off

First of all thx for the library, is super simple to implement and seems to do exactly what I need. I'm just running into a couple "problems" that I believe are related to the hardware that I'm using and not sure if the problem is related to the library itself.

My setup:
- Raspberry Pi Zero W
- Adafruit LED ring x12
- Raspbian
- Running Python

My Problem:
If I run the demo, or any of the functions and try set all the leds on it works ,some times I need to run the function multiple times because not all leds would apply the new color setting, nothing major here. I can live with this little problem. The real problem comes when I try to set the all to off, there is 1 led that will always stay on no matter what. (Seems to be the first led).

Has anyone runned into this problem before?

I have tried the PWM and PCM, on different GPIO but behaviour perists. :(

penfold42, 2018-03-16:
What level shifter are you using ?

A quick hack is to drop the 5v supply to the leds using a 1 amp diode in the power line

umbertix, 2018-03-16:
I'm powering my ring directly from a microusb connector (splitting ground and pwr), so direct 5v to it.

If you think that adding a 1amp diode will fix this I will try get my hands on one. Sorry my electronic knowledge is basically `null`. Thx for the input.

eloots, 2018-03-16:
Powering a strip with 5 Volt and driving it with a 3.3V logical signal is a problem. The issue though is that the problem(s) may not be obvious at first as some effects are spurious. I’ve successfully applied the diode solution (using an 8-LED Neopixel strip).

As was pointed out in other discussions, this solution is only practical when driving a relatively low number of LEDs. Each LED consumes max. 60mA, so, for a 1A diode, that corresponds to ±16 diodes.

umbertix, 2018-03-16:
So if I'm using 12 led, 1A will be enough? Or shall I go for a bit bigger? (Sorry for the nooby questions)
I was looking at this [one](https://www.maplin.co.uk/p/1n4001s-1a-silicon-rectifier-ql73q). Thx for the help guys

eloots, 2018-03-16:
Yes that should be ok. I’m also using the 1N4001 you plan to use.

penfold42, 2018-03-16:
It will be ok

12 leds, .06 amp each at full white
= 0.72 Amps maximum current.

umbertix, 2018-03-16:
Unfortunetly I seem to have something not quite correct. The Led still light up at all time even with the diode on the power line. I have tried with another LED ring just to discard a malfunction on the ring.

Just out of curiosity is anyone else running it on a Raspberry Pi Zero W? If so with what config (pin, Mhz)?

closed 2018-03-17 by umbertix