jgarff/rpi_ws281x #281: The LEDs are flickering randomly

CosminZoRR, 2018-03-22:
Hello!

I know there are many inssues with the same title, but no one could help me.

I am making a robot and I want to play some sound and also have some light effects. I know I can't have both using PWM so I tried to use PCM or SPI. 

Using PWM everything is working good, but if I switch the PIN to GPIO 21 and LED_CHANNEL to 0 to use the PCM I get some random colors and the LEDs are flickering randomly.

Can someone help me with this, please? 

Thanks in advance!

Gadgetoid, 2018-03-27:
What LEDs/LED board are you using?

Are you sure you're switching them over to the right pin when you change your code?

CosminZoRR, 2018-03-27:
Hi! Thanks for the reply.

I am using this board: [BOARD](https://www.optimusdigital.ro/ro/optoelectronice-altele/749-inel-de-led-uri-rgb-ws2812-cu-16-led-uri.html)

Yes, I am sure I am using the correct PINs

willgower, 2018-09-14:
Make sure the ground of the rpi/microcontroller is connected to the ground of the leds

FrankX0, 2018-09-21:
My guess would be noisy power supply.
Try a different power supply or add extra decoupling.

AngCosmin, 2018-09-21:
Yea. Everything was good. I don’t have the project anymore to test stuff, but thanks anyway. We can close the issue.

closed 2018-09-21 by AngCosmin