jgarff/rpi_ws281x #283: Stepped brightness.

rnlgreen, 2018-03-29:
Hi,
I'm using adafruit neopixel sticks, connected to GND, 5V and GPIO18 on my Raspberry Pi 3B.   I can run strandtest.py fine (as root), but 've noticed the colour/brightness work in steps.  If I try and set the RGB colour to 23, 23, 23, the LEDs stay off.... 24 is the minimum value to illicit light.

The brightness steps up at r/g/b values of 24, 47, 70, 94.  ... etc. (gets harder to tell exactly when it steps as it gets brighter).

I've used identical code to control the same LEDs through pimorini's pantilt-hat and associated python library and the LEDs work smoothly all the way from 1 to 255, just changing the API calls I use to set the pixels.

Is there some limitation to this library or the way the PI controls them directly, vs. whatever method pimoroni pantilt-hat is using?  Or am I missing something else?

Thanks,!

rnlgreen, 2018-03-29:
... this also means that at lower brightness colours are badly wrong.  rgb 23,23,24 is pure blue (red and green are off still), rgb 24,24,25 is white (all colours are on).

penfold42, 2018-03-29:
Weird - I’ve not seen that behaviour before.

I’m sure i could reproduce it with a (weird) custom gamma table

jgarff, 2018-03-29:
Yeah, I not seen this behavior either.  On the LEDs I have, the values go
smoothly from 0 to 255.

rnlgreen, 2018-03-29:
Ah.... could be self inflicted.  Sorry to disturb you!

I had LED_BRIGHTNESS set to 10... hadn't consciously set the brightness elsewhere in the code so didn't twig it was being initialised low.

closed 2018-03-29 by rnlgreen

jgarff, 2018-03-29:
No problem, glad you figured it out.

penfold42, 2018-03-29:
And we have it captured for someone in a similar position in the future