jgarff/rpi_ws281x #284: Segmentation fault when calling ws2811_fini

natcl, 2018-03-30:
There seems to be a segmentation fault when calling ws2811_fini in the nodejs bindings.
Any tips on how to fix that ?

Thanks !

```
void finalize(const Nan::FunctionCallbackInfo<v8::Value> &info)
{
  ws2811_return_t ret;

  ret = ws2811_wait(&ws281x);
  if (ret != WS2811_SUCCESS)
  {
    Nan::ThrowError(ws2811_get_return_t_str(ret));
    return;
  }

  ws2811_fini(&ws281x);
}
```

danielytics, 2018-04-04:
I’ve noticed the same thing a few days ago, while using C++ (but not while using the included test program, which is odd because my code was based off it).

etanx, 2018-08-01:
This may be slightly different, but when I run any of the python example scripts I get:

`File "build/bdist.linux-armv6l/egg/neopixel.py", line 102, in begin
RuntimeError: ws2811_init failed with code -1
Segmentation fault`

I could get an older x16 RGB LED ring to work, but now with a x24 RGBW (I used a branch sk6812) it's not initializing.

davthomaspilot, 2018-08-03:
Check out the issue I just posted.  Try intializing .chan[0].gamma and chan[1].gamma to zero.