jgarff/rpi_ws281x #286: Support for neopixel RGBW rings

walkerdb, 2018-04-08:
The extra "white" LED in each pixel on the RGBW neopixel rings seem to be throwing off LED indexing on the python bindings.

## Expected behavior:
Calling `setPixelColor` on each pixel index with `11111111` for each color should set the RGB values of that pixel to max brightness.

## Actual behavior
Index 0 correctly sets RGB of the first pixel
Index 1 sets the white LED of the first pixel along with the RG LEDs of the second.
Index 2 sets BW of the second pixel and only G of the third

etc.

The following code replicates this behavior:

```python
strip.begin
strip = Adafruit_NeoPixel(16, 18, 800000, 10, False, 10, 0)
strip.begin()

green = '11111111'
red = '11111111'
blue = '11111111'
color = int(green + red + blue, 2)

strip.setPixelColor(0, color)  # RGB pixel 1
strip.setPixelColor(1, color)  # W pixel 1, RG pixel 2
strip.setPixelColor(2, color)  # BW pixel 2, G pixel 3
strip.show
```

Currently running on an rPi Zero WH with NeoPixel Ring `16 x 5050 RGBW` LEDs (datasheet [here](https://cdn-shop.adafruit.com/datasheets/WS2812.pdf))

If I had to venture a guess it looks a lot like an off-by-one error somewhere, almost like it's missing an extra byte for each pixel.

penfold42, 2018-04-08:
I didn’t see you set the strip type

https://github.com/jgarff/rpi_ws281x/blob/master/python/examples/SK6812_strandtest.py

walkerdb, 2018-04-08:
That was 100% it. Didn't realize that was a feature!

Thanks for the help.

closed 2018-04-08 by walkerdb