jgarff/rpi_ws281x #289: Got a typecast error while compile

ca-haddock, 2018-04-17:
Hi 
while compiing i got the following error:

rpi_ws281x]# scons
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
CC      mailbox.o
mailbox.c: In function 'unmapmem':
mailbox.c:71:25: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
     uint32_t baseaddr = (uint32_t)addr & pagemask;
                         ^
mailbox.c:74:16: error: cast to pointer from integer of different size [-Werror=int-to-pointer-cast]
     s = munmap((void *)baseaddr, size);
                ^
cc1: all warnings being treated as errors
scons: *** [mailbox.o] Error 1
scons: building terminated because of errors.
Output of gcc -v 

 gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/aarch64-unknown-linux-gnu/7.3.1/lto-wrapper
Target: aarch64-unknown-linux-gnu
Configured with: /build/gcc/src/gcc/configure --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://github.com/archlinuxarm/PKGBUILDs/issues --enable-languages=c,c++,fortran,go,lto,objc,obj-c++ --enable-shared --enable-threads=posix --with-system-zlib --with-isl --enable-__cxa_atexit --disable-libunwind-exceptions --enable-clocale=gnu --disable-libstdcxx-pch --disable-libssp --enable-gnu-unique-object --enable-linker-build-id --enable-lto --enable-plugin --enable-install-libiberty --with-linker-hash-style=gnu --enable-gnu-indirect-function --disable-multilib --disable-werror --enable-checking=release --enable-default-pie --enable-default-ssp --host=aarch64-unknown-linux-gnu --build=aarch64-unknown-linux-gnu --with-arch=armv8-a --enable-fix-cortex-a53-835769 --enable-fix-cortex-a53-843419
Thread model: posix
gcc version 7.3.1 20180312 (GCC)

Gadgetoid, 2018-04-19:
For what it's worth, it builds okay for me on gcc 6.3.0, directly on Raspbian. I think it's largely untested and unsupported on Arch Linux, and suspect this will be the first of many problems. Interested to know if you get any further, though, since there would be no reason not to attempt to fix these issues if they don't break Raspbian compat.

Gadgetoid, 2018-04-24:
FWIW I managed to get `rpi-ws281x-python` (https://github.com/rpi-ws281x/rpi-ws281x-python) to build using this patch:

```patch
diff --git a/library/Makefile b/library/Makefile
index b8da8f5..df3df72 100644
--- a/library/Makefile
+++ b/library/Makefile
@@ -14,7 +14,7 @@ $(BUILD)/version.h:
        cp version.h $(BUILD)/version.h

 $(OBJECTS): $(BUILD)/%.o : $(SRC)/%.c
-       gcc $< -o $@ -c -g -O2 -Wall -Werror -fPIC
+       gcc $< -o $@ -c -g -O2 -Wall -Werror -Wno-int-to-pointer-cast -Wno-pointer-to-int-cast -fPIC

 $(BUILD)/$(LIB): $(OBJECTS)
        ar rc $@ $^
```

However, the resulting library *still wont work*:

```
[root@alarm examples]# python hat/matrix.py
Traceback (most recent call last):
  File "hat/matrix.py", line 6, in <module>
    import unicornhat as unicorn
  File "/usr/lib/python3.6/site-packages/unicornhat.py", line 35, in <module>
    ws2812.begin()
  File "/usr/lib/python3.6/site-packages/rpi_ws281x-3.1.0-py3.6-linux-aarch64.egg/rpi_ws281x/rpi_ws281x.py", line 124, in begin
    raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, str_resp))
RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)
```

This is because `rpihw.c` relies upon the output of `/proc/cpuinfo`:

https://github.com/jgarff/rpi_ws281x/blob/e4a05d6538c02bb9714f2efc6630f2bfdcf35bf6/rpihw.c#L327

And it's specifically looking for a line starting with `"Revision:"` which is not present on AArch64:

```text
[root@alarm examples]# cat /proc/cpuinfo
processor       : 0
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4
```

But is present on Raspbian:

```
pi@raspberry:~ $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a22082
Serial          : 000000008818b4ec
```

Presumably this is because AArch64 has a completely mainline kernel and none of the Raspberry Pi specific hacks. This results in hardware, version and serial information not being present in `/proc/cpuinfo`.

This information - or at least a relevant portion of it - is, however, available via the device-tree so perhaps an additional check there might suffice:

on AArch64:

```
[alarm@alarm ~]$ cat /proc/device-tree/model
Raspberry Pi 3 Model B
```

So with that check short circuited by simply forcefully returning a result in `rpihw.c/rpi_hw_detect()` we now run into:

```
[root@alarm library]# python ../../unicorn-hat/examples/hat/matrix.py
Can't open device file
: Permission denied
Traceback (most recent call last):
  File "../../unicorn-hat/examples/hat/matrix.py", line 6, in <module>
    import unicornhat as unicorn
  File "/usr/lib/python3.6/site-packages/unicornhat.py", line 35, in <module>
    ws2812.begin()
  File "/usr/lib/python3.6/site-packages/rpi_ws281x-3.1.0-py3.6-linux-aarch64.egg/rpi_ws281x/rpi_ws281x.py", line 124, in begin
    raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, str_resp))
RuntimeError: ws2811_init failed with code -9 (Failed to create mailbox device)
```

Which is, presumably, the point at which we're trying to access the Videocore mailbox? Which might be a problem since AArch64 states: "Use this installation only if you have no dependencies on the closed source vendor libraries shipped in the ARMv7 release." I'm guessing this constitutes a dependency? I'm a bit out of my depth here.

penfold42, 2018-04-24:
Spi mode has more chance of working - no mailbox dma requirements

Gadgetoid, 2018-04-25:
That's a point- I've archived the SD card I ran through this process on, so I should be able to pick it up and try SPI when I've got a moment to spare. It doesn't help Unicorn HAT users, but if it works at all then it might be worth patching `rpihw.c` and Scons to allow it to at least build/run.

Although the pointer<->int casting errors highlighted by gcc 7.x.x should probably be fixed?

commit a8441f67 (2018-11-12): Fixes for arm64 support (#316)

* Fix compilation on arm64 (#173, #289)

* Read CPU revision from /proc/device-tree on arm64 (closes #289)

* rpihw: move declarations to the top of rpi_hw_detect()
closed 2018-11-12 by jgarff