jgarff/rpi_ws281x #290: BadZipfile, "File is not a zip file"

willcode4cookiedough, 2018-04-19:
I have a fresh install of Rasbian stretch lite that is updated and upgraded. 

I installed the dependencies and complied following instructions and can verify that setuptools-5.7.zip is downloaded to the Python folder in the cloned repo. 

Any insight would be greatly appreciated. Thank you

Extracting in /tmp/tmpejd0zm
Traceback (most recent call last):
  File "./setup.py", line 4, in <module>
    use_setuptools()
  File "/home/pi/libs/rpi_ws281x/python/ez_setup.py", line 140, in use_setuptools
    return _do_download(version, download_base, to_dir, download_delay)
  File "/home/pi/libs/rpi_ws281x/python/ez_setup.py", line 120, in _do_download
    _build_egg(egg, archive, to_dir)
  File "/home/pi/libs/rpi_ws281x/python/ez_setup.py", line 62, in _build_egg
    with archive_context(archive_filename):
  File "/usr/lib/python2.7/contextlib.py", line 17, in __enter__
    return self.gen.next()
  File "/home/pi/libs/rpi_ws281x/python/ez_setup.py", line 100, in archive_context
    with ContextualZipFile(filename) as archive:
  File "/home/pi/libs/rpi_ws281x/python/ez_setup.py", line 88, in __new__
    return zipfile.ZipFile(*args, **kwargs)
  File "/usr/lib/python2.7/zipfile.py", line 770, in __init__
    self._RealGetContents()
  File "/usr/lib/python2.7/zipfile.py", line 813, in _RealGetContents
    raise BadZipfile, "File is not a zip file"
zipfile.BadZipfile: File is not a zip file

Gadgetoid, 2018-04-20:
If you need the Python library, just `sudo pip install rpi_ws281x`

JulioBBL, 2018-04-26:
I have the same issue, tried instaling through pip with no luck. Every time I try to run my code I get a message that there is no module named 'neopixel'.

Any other method I should try?

hwassie, 2018-04-26:
Try:
sudo apt-get install python-pip
and then 
 python ./setup.py build
again

hnoesekabel, 2018-05-06:
Looking at the downloaded setuptools-5.7.zip I saw only the redirect response from the webserver... it was no binary ZIP file, so the error message was correct.

Worked for me: delete the downloaded setuptools-5.7.zip and manually fetch it with wget or curl and then run setup again...
```
wget https://pypi.python.org/packages/source/s/setuptools/setuptools-5.7.zip
python ./setup.py build
sudo python ./setup.py install
```

iknowcss, 2018-05-19:
@hnoesekabel had the same issue and that worked for me, cheers!

Ruru-embodied, 2018-06-13:
@hnoesekabel Thanks! This worked for me.  Could I ask how to figure out the setuptools-5.7.zip is a redirect response from the webserver, not binary zip file.

gilvad, 2018-06-25:
for me it works when I use pip3 and not pip to install it. Then using python3 to execute the scripts.

etanx, 2018-07-13:
Using pip did not work for me. I could install with python3 but I wanted it to run it with python2 like all the others. This manual fetch with wget worked for me!

john-santose, 2018-07-25:
This resolved the issue that I was having as well and should be placed as a fix on the main github page for the library. Thanks for finding this workaround.

Gadgetoid, 2018-08-28:
Note: the Python bindings left over in this repository are obsolete, you should use: https://github.com/rpi-ws281x/rpi-ws281x-python

Or: `sudo pip install rpi_ws281x`

You should also use `rpi_ws281x` instead of `neopixel`.

Any issues with installing/using Python packages should be raised here: https://github.com/rpi-ws281x/rpi-ws281x-python/issues