jgarff/rpi_ws281x #292: Sk6812-Rgbw RPi2

Soupofdoom, 2018-04-24:
I'm trying to run a SK6812RGBW strip (30/m x 1m) from a RPi2 B+ and no matter what I try when I output to the LEDs I get the pattern red, green, blue and off across the strip when attempting a singular color wipe to blue. The pattern shifts along when I try to wipe to a different color. 

I looked at penfold42's branch from a while back (and even tried that) but nothing helped. 

I changed the strip type in the config and disabled audio PWM and forced HD I audio. 

 LED_STRIP      = ws.SK6812_STRIP_RGBW

I'm guessing that this is a timing issue and/or I'm just missing something essential/being retarded. 

Can anyone help?

penfold42, 2018-04-24:
What if you set the strip type to ws2812 ?

I wonder if they sent you an sk6812RGB (no white) strip

Soupofdoom, 2018-04-24:
So I haven't seen it myself (I'm at work) but I talked the wife through testing this and it does seem that this is the case.

Ill update later to confirm.

Soupofdoom, 2018-04-24:
Okay, so just got home and this is indeed the case! Thanks for the suggestion @penfold42!

Time to get complaining to that company!

closed 2018-04-24 by Soupofdoom

penfold42, 2018-04-25:
Don’t let them whitewash the issue