jgarff/rpi_ws281x #294: error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1 on RPi 3b when 'python setup.py install'

JayJoeSeventh, 2018-04-30:
I wanted to 'import neopixel' from anywhere, so I try to do the installation as shown. but it fails with error 'arm-linux-gnueabihf-gcc'.

Help! :(

eloyasensio, 2018-05-09:
Hi!
I had same problem.
But I tried to follow the steps of this https://learn.adafruit.com/neopixels-on-raspberry-pi  and then I can build this without problem.
I think that solution is install "scons":
sudo apt-get update
sudo apt-get install build-essential python-dev git scons swig

FuriosoJack, 2018-08-23:
For those that have not worked at all you have to install these packages 

`sudo apt-get install libzbar-dev libzbar0`

KEINOS, 2018-11-02:
I think you need to install python-dev too. In order to let gcc compile the Python extensions.

```
sudo apt install python-dev
```

Sorry if it's not the case.

tonytanjk, 2019-01-03:
I tried everything but it’s a fail

tonytanjk, 2019-01-05:
sudo apt-get install portaudio19-dev.
Try this

chrisdimoff, 2019-03-22:
1) Ensure SPI is enabled in `raspi_config`
2) `cd rpi_ws2811x` && run `scons`
3) try again
4) be happy?

tlilianas, 2019-07-31:
this is what you are looking for guys : 
sudo apt install -y python python-pip libffi-dev python-backports.ssl-match-hostname
sudo pip install docker-compose

jvanlangen, 2019-09-20:
Too bad, still:
 Building wheel for cryptography (PEP 517) ... error

erm3nda, 2019-12-12:
@jvanlangen Your comment is insufficient.
In most of those cases, the problem comes when building a package that is required by the main package. So, you should be able to see which one is the offending package in the terminal then fix installing it's package-dev or whatever it needs.

Becase `arm-linux-gnueabihf-gcc` is a compiler, it will need always the dev packages, that includes headers for let gcc compile it.

So, run again your command, see what's the last package you can identify in the terminal, then come back.

erm3nda, 2019-12-12:
@jvanlangen 

My fail, you refer to `cryptography` package.
Take a look at SO thread -> https://raspberrypi.stackexchange.com/questions/62364/installing-cryptography-using-pip-on-raspbian-jessie-image

pawKer, 2019-12-12:
Just putting this out here in case it might help someone, I was getting this error because I hadn't run the `scons` command in the `rpi_ws281x` folder before running `python setup.py build` and `python setup.py install`.

pawKer, 2019-12-12:
> Just putting this out here in case it might help someone, I was getting this error because I hadn't run the `scons` command in the `rpi_ws281x` folder before running `python setup.py build` and `python setup.py install`. Thanks @chrisdimoff !

pawKer, 2019-12-12:
> 1. Ensure SPI is enabled in `raspi_config`
> 2. `cd rpi_ws2811x` && run `scons`
> 3. try again
> 4. be happy?

Just putting this out here in case it might help someone, this fixed it for me. I was getting the same error because I hadn't run the `scons` command in the `rpi_ws281x` folder before running `python setup.py build` and `python setup.py install`. Exactly like @chrisdimoff says above.

maulalwi, 2020-06-16:
sudo apt install clang -y

Fixed.

rahulshyokand, 2020-06-28:
### These Commands Fixed Similar Issue on my Rpi 3b
Note: These installations may not have any problem because some are already installed and some are not.
sudo python3 -m pip install --upgrade pip
sudo python3 -m pip install pyparsing==2.1.0
sudo python3 -m pip install pyparser==1.0
sudo apt-get install libffi-dev
sudo apt-get install libzbar-dev
sudo apt-get install clang -y

sudo apt-get install python3-dev
 sudo apt-get install libssl-dev
### Problem seems to be in cffi package
**Following command also installs cffi package which will test installation**
sudo python3 -m pip install cryptography==2.2.2
Alternativly you can try 
sudo python3 -m pip install cffi
### Here is the potential Error which gets solved 
The issue in includes of openssl.h and python.h were solved
 <#include 'Python.h'>
<#include 'openssl.h'> 
[image]

maulalwi, 2020-07-08:
also met again with the same thing. then searching and solving

```shell
sudo apt --reinstall install libc6 libc6-dev
```

jsxvenom, 2020-07-16:
@rahulshyokand  thank you 
I am a newbie for programming, I have same problem when try to "pip install pycurl" and show the same error.
But the problem had solved using your list of installation that you mention above.
I am new also to this github.. I dont know to quote your post.. 
thanks

Gadgetoid, 2020-07-16:
The Python code in this repository is hopelessly obsolete, you should use the shipped Python library, source here: https://github.com/rpi-ws281x/rpi-ws281x-python

```
sudo pip install rpi_ws281x
```

closed 2020-07-16 by Gadgetoid

datty258, 2020-09-08:
> For those that have not worked at all you have to install these packages
> 
> `sudo apt-get install libzbar-dev libzbar0`

work for me

ku113p, 2020-09-13:
worked for me
`sudo apt install -y libffi-dev`

webCoderLOL, 2020-12-05:
@maulalwi Thanks! This works for me when installing rpi.gpio over Python3 virtual env.
`sudo apt install clang -y`

manknewl, 2021-04-04:
This fixed it for me: 

sudo python3 -m pip install --upgrade pip
sudo python3 -m pip install pyparsing==2.1.0
sudo python3 -m pip install pyparser==1.0
sudo apt-get install libffi-dev
sudo apt-get install libzbar-dev
sudo apt-get install clang -y

JulienChapron, 2021-04-13:
> This fixed it for me:
> 
> sudo python3 -m pip install --upgrade pip
> sudo python3 -m pip install pyparsing==2.1.0
> sudo python3 -m pip install pyparser==1.0
> sudo apt-get install libffi-dev
> sudo apt-get install libzbar-dev
> sudo apt-get install clang -y

and `sudo apt-get install python3-dev` on raspberry pi 4

dmarcelino, 2021-07-29:
For me, raspberry pi 4, it was:

> sudo python3 -m pip install --upgrade pip
> sudo python3 -m pip install pyparsing==2.1.0
> sudo python3 -m pip install pyparser==1.0
> sudo apt-get install libffi-dev
> sudo apt-get install libzbar-dev
> sudo apt-get install clang -y
> and sudo apt-get install python3-dev on raspberry pi 4

Plus `sudo apt-get install build-essential`

NuwanMadhusanka0417, 2022-02-17:
if your error shows "fatal error: python.h: no such file or directory completion terminated", you have to try this one. this fix my problem

https://stackoverflow.com/questions/21530577/fatal-error-python-h-no-such-file-or-directory