jgarff/rpi_ws281x #296: RuntimeError: ws2811_init failed with code -5 (mmap() failed)

melzoghbi, 2018-06-01:
Hi,

I am getting this error after i installed neopixel on my RPI3 model B. This was working before i did:
apt-get upgrade 

But after i re-installed my conda environment with all required packages including rpi_ws281x i am still getting this error:

RuntimeError: ws2811_init failed with code -5 (mmap() failed)

Here is my current version of Raspbian stretch:

Linux rpi_moelzogh 4.14.34-v7+ #1110 SMP Mon Apr 16 15:18:51 BST 2018 armv7l GNU/Linux

Any advice?

Thanks.

nvl1109, 2018-06-02:
I'm using RPi Zero W, and got same issue.
About 1month ago, I install this and make WS2812B led strip works fine.
From that time, I don't upgrade the system.

I'm using PWM mode.

melzoghbi, 2018-06-07:
@nvl1109 i do have root permission and i am having this issue.

Gadgetoid, 2018-06-12:
What happens if you run `sudo python strandtest.py`?

closed 2018-08-29 by Gadgetoid