jgarff/rpi_ws281x #297: Error when installing with pip

charliesneath, 2018-06-02:
I'm running `pip install rpi_ws281x` and seeing the following error: 

```
Collecting rpi_ws281x
  Downloading https://files.pythonhosted.org/packages/20/63/8e6ee92120b556dec3ee0b28e3919e6d61cee54bb9438b9e18ae9f6558b3/rpi_ws281x-3.1.0.tar.gz (59kB)
    100% |████████████████████████████████| 61kB 1.0MB/s 
Installing collected packages: rpi-ws281x
  Running setup.py install for rpi-ws281x ... error
    Complete output from command /Users/charliesneath/.pyenv/versions/3.6.5/bin/python3.6 -u -c "import setuptools, tokenize;__file__='/private/tmp/pip-install-1_0tijtd/rpi-ws281x/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /private/tmp/pip-record-cw231f1s/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_py
    Compiling ws281x library...
    b"lib/ws2811.c:41:10: fatal error: 'linux/types.h' file not found\n#include <linux/types.h>\n         ^~~~~~~~~~~~~~~\n1 error generated.\nmake: *** [lib-built/ws2811.o] Error 1\n"
    creating build
    creating build/lib.macosx-10.13-x86_64-3.6
    creating build/lib.macosx-10.13-x86_64-3.6/neopixel
    copying neopixel/__init__.py -> build/lib.macosx-10.13-x86_64-3.6/neopixel
    creating build/lib.macosx-10.13-x86_64-3.6/rpi_ws281x
    copying rpi_ws281x/rpi_ws281x.py -> build/lib.macosx-10.13-x86_64-3.6/rpi_ws281x
    copying rpi_ws281x/__init__.py -> build/lib.macosx-10.13-x86_64-3.6/rpi_ws281x
    running build_ext
    building '_rpi_ws281x' extension
    creating build/temp.macosx-10.13-x86_64-3.6
    clang -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -Ilib/ -I/Users/charliesneath/.pyenv/versions/3.6.5/include/python3.6m -c rpi_ws281x_wrap.c -o build/temp.macosx-10.13-x86_64-3.6/rpi_ws281x_wrap.o
    clang -bundle -undefined dynamic_lookup -L/usr/local/opt/readline/lib -L/usr/local/opt/readline/lib -L/usr/local/opt/openssl/lib -L/Users/charliesneath/.pyenv/versions/3.6.5/lib -L/usr/local/opt/readline/lib -L/usr/local/opt/readline/lib -L/usr/local/opt/openssl/lib -L/Users/charliesneath/.pyenv/versions/3.6.5/lib build/temp.macosx-10.13-x86_64-3.6/rpi_ws281x_wrap.o -Llib-built/ -lws2811 -o build/lib.macosx-10.13-x86_64-3.6/_rpi_ws281x.cpython-36m-darwin.so
    ld: library not found for -lws2811
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    error: command 'clang' failed with exit status 1
    
    ----------------------------------------
Command "/Users/charliesneath/.pyenv/versions/3.6.5/bin/python3.6 -u -c "import setuptools, tokenize;__file__='/private/tmp/pip-install-1_0tijtd/rpi-ws281x/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /private/tmp/pip-record-cw231f1s/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/tmp/pip-install-1_0tijtd/rpi-ws281x/
```

Gadgetoid, 2018-06-04:
This will not work on OSX, and I'm not entirely sure:

1. It could ever work reasonably
2. It would be worth even trying

Since rpi_ws281x relies on *very* specific code that runs only on the Raspberry Pi, and also links various linux headers, it's just not cross-platform compatible in any meaningful way. Even if you could install it locally, it wouldn't function. 

What's the standard procedure in cases like this with Python libraries? Is there a standard procedure? We have a library targeted at one specific platform that's openly available on pip- should it recognise and stub out the various methods and features when installed on another platform? Should it raise an error blocking the install/compilation?

charliesneath, 2018-06-07:
Thanks for clarifying this. I'm running this on my Raspberry Pi and won't worry about installing on the Mac.

closed 2018-06-07 by charliesneath