jgarff/rpi_ws281x #299: Getting "zipfile.BadZipfile: File is not a zip file" when running Python build

charliesneath, 2018-06-11:
I've just flashed my Raspberry Pi memory card and have a fresh install of this repo. 

When running `python rpi_ws281x/python/setup.py build`, I'm getting the following error.

```
Downloading https://pypi.python.org/packages/source/s/setuptools/setuptools-5.7.zip
Extracting in /tmp/tmpym57pu
Traceback (most recent call last):
  File "rpi_ws281x/python/setup.py", line 4, in <module>
    use_setuptools()
  File "/home/pi/weatherbox/rpi_ws281x/python/ez_setup.py", line 140, in use_setuptools
    return _do_download(version, download_base, to_dir, download_delay)
  File "/home/pi/weatherbox/rpi_ws281x/python/ez_setup.py", line 120, in _do_download
    _build_egg(egg, archive, to_dir)
  File "/home/pi/weatherbox/rpi_ws281x/python/ez_setup.py", line 62, in _build_egg
    with archive_context(archive_filename):
  File "/usr/lib/python2.7/contextlib.py", line 17, in __enter__
    return self.gen.next()
  File "/home/pi/weatherbox/rpi_ws281x/python/ez_setup.py", line 100, in archive_context
    with ContextualZipFile(filename) as archive:
  File "/home/pi/weatherbox/rpi_ws281x/python/ez_setup.py", line 88, in __new__
    return zipfile.ZipFile(*args, **kwargs)
  File "/usr/lib/python2.7/zipfile.py", line 770, in __init__
    self._RealGetContents()
  File "/usr/lib/python2.7/zipfile.py", line 813, in _RealGetContents
    raise BadZipfile, "File is not a zip file"
zipfile.BadZipfile: File is not a zip file
```

charliesneath, 2018-06-11:
Saw that this is duplicate of #290. Please close if you're tracking this elsewhere.

Will start trying Python library via `sudo pip install rpi_ws281x` for now.

Gadgetoid, 2018-06-12:
Yes it's my intent to move for the EOL'ing of the Python bindings in this library, in favour of the separately maintained, pip'd alternatives.

closed 2018-06-12 by Gadgetoid