jgarff/rpi_ws281x #300: Running SK6812 RGBW

Dr-Zee, 2018-06-23:
When running SK6812_strandtest.py (or any function trying to pass a white value).
The script runs until it hits a 4 value color

File "python/examples/SK6812_strandtest.py", line 94, in <module>
    colorWipe(strip, Color(0, 0, 0, 255))  # White wipe
TypeError: Color() takes exactly 3 arguments (4 given)

I set the led count and strand type in main.c, and built with scons, then ran setup.py in the python folder.

Dr-Zee, 2018-07-11:
Found I was using the wrong branch.
https://github.com/penfold42/rpi_ws281x was what I was looking for.

closed 2018-07-11 by Dr-Zee

Dr-Zee, 2018-07-19:
I can double-check later this week. I destroyed, pulled and initialized the main branch several times and every time got the 4 argument error.
Not saying you're wrong, but that was my experience. Could easily have been user error.

ShakataGaNai, 2018-11-05:
So I've installed the repo via `pip install rpi_ws281x` and `pip3 install rpi_ws281x` and tried to run both the sk6812 strand test, along with white test files. Both fail:

```
Traceback (most recent call last):
  File "SK6812_white_test.py", line 47, in <module>
    colorWipe(strip, Color(0, 0, 0, 255), 0)  # White wipe
TypeError: Color() takes exactly 3 arguments (4 given)
```
(Note: I updated the `from neopixel import *` to `from rpi_ws281x import *` )