jgarff/rpi_ws281x #301: SK6812 showing random colors and flickering

cshapeshifter, 2018-07-21:
- [This LED strip](https://www.aliexpress.com/item/Wholesale-4m-5m-SK6812-similar-ws2812b-RGBW-4-in-1-Waterproof-IP65-IP67-60-leds-pixles/32476317187.html?spm=a2g0s.9042311.0.0.27424c4dI12uA1) from AliExpress
- Raspberry Pi Zero W
- PWM on GPIO 18 (but I tried SPI on GPIO 10 and PCM on GPIO 21 with basically the same results)
- ```blacklist snd_bcm2835```
- ```/boot/config.txt```: [http://ix.io/1i2c](http://ix.io/1i2c)

When I run ```python examples/SK6812_white_test.py``` the LEDs show activity but it seems very random. Here's what the first three iterations (colorWipe for R, G and B) look like:
[image]

- Most of the time the W LEDs are on and overwhelming
- Brightness settings seem to have no effect
- Colors are all wrong and random
- There's is some flicker, especially on the first few LEDs when no new data is supplied.

What I tried:
- Messing around with FREQ_HZ
- Trying different SK6812 types, like _STRIP_BRGW etc. (same but different results)
- Different strip
- Different GPIO pins.

I have no idea what is wrong.

cshapeshifter, 2018-07-22:
Turns out this is what happens when you forget to connect the level shifter GND to the pi's GPIO GND. Problem solved.

closed 2018-07-22 by cshapeshifter

patricker, 2020-10-24:
Two years later, and your comment saved me! Thanks alot @cshapeshifter.  I had not hooked up my LED light strips +/- to my Raspberry Pi since I had an external power source.  The moment I connected up the ground all the colors came out perfect.

tobiaswaelde, 2022-06-11:
I am currently building a world map. Just placing the LEDs was like hell already. But as it came to testing the LEDs the pure hell began. I wrote a script, simple as it could be, looping throug all the LEDs. Every time, after exactly 47 working LEDs, all LEDs started blinking in any color like in video from you. Had no level shifter that time. Chip is a ESP32.

I figured out that some parts of my LED strips are not connected to GND in like 4 places. It took me 3 days now to figure that out :)

[image]
 PS: The glass pane is just for testing and as template to drill the holes in the wood later.

Sirllamaa, 2025-04-28:
> Two years later, and your comment saved me! Thanks alot @cshapeshifter.  I had not hooked up my LED light strips +/- to my Raspberry Pi since I had an external power source.  The moment I connected up the ground all the colors came out perfect.
> 
> 

And again 5 years later :)