jgarff/rpi_ws281x #302: WS2812B Strip not lighting up with Raspi 2 Model B

Vioxe, 2018-07-24:
I have a Raspi 2, heres all the wiring:
https://drive.google.com/file/d/1fuZobCAEKu5-_xNCtb_3BH2-0b3GgtVJ/view?usp=sharing 
I have a 74AHCT125 logic level shifter and i have the Pi GPIO pin 18 going to 1A on the IC and i have the DIN on the strip going to 1Y on the IC. This should be correctly wired as i followed Adafruits wiring guide. i have the library installed and the strandtest.py file edited to my needs. Im not the best with lunix but i just followed guide to get all the software side done and when i do: "sudo python strandtest.py" it will say the annimations it is doing but nothing comes on on my led strip. Any help would be greatly appreciated. Thanks!

Vioxe, 2018-07-24:
Could this be because the library doesn't support Raspi 2?

BruceMardle, 2018-07-25:
I first tried using the PWM0 interface on my Pi 2 (i.e. BCM pin 18, which is physical pin 12). It didn't work, so I tried the SPI interface, which did work.I later read that the Pi 2's audio uses PWM0. You can probably disable that; I don't know; I'm happy using SPI.

   

https://drive.google.com/file/d/1fuZobCAEKu5-_xNCtb_3BH2-0b3GgtVJ/view?usp=sharing

Vioxe, 2018-07-25:
Okay I want to try that. I'm not particularly sure how though. I'm assuming you would just change the pin to SPI in the strandtest.py correct? I'll try that and see if it works. Thanks!

BruceMardle, 2018-07-25:
Yes, connect the first LED's Din to the Pi's SPI0 MOSI (master out, serial in; physical pin 19). I can't remember if you have to tell the software to use SPI. Maybe it figures it out from which pin it's using.

BruceMardle, 2018-07-25:
... and when I say "connect the first LED's Din to the Pi's SPI0 MOSI", I mean via a level-shifter (unless you've got a diode in series with the LEDs' power supply)! But you knew that already :-)

Vioxe, 2018-07-25:
Some of my thoughts:
-The version of Debian stretch lite I'm using is the newest and because of that the library doesn't work on it?
-The led strip I'm using is faulty? 
-The Pi I'm using is faulty? 
-The power supply is faulty? 
-The wiring is not correct? 
-The logic level shifter is bad?
-There is too much noise on the DIN line? 
Those are all the possible faults I could think of.

Vioxe, 2018-07-25:
Could you maybe tell me your exact setup?

BruceMardle, 2018-07-25:
I don't have my Pi with me at the moment. It's running Jessie.

Vioxe, 2018-07-26:
Alright Ill try jessie out and see if that works.