jgarff/rpi_ws281x #304: /dev/gpiomem

anfractuosity, 2018-07-26:
Hi,

First of all thanks a lot for your work on this project, I've used it for a number 
of projects! ( Most recently - https://www.anfractuosity.com/projects/painting-a-christmas-tree/ )

I'm just wondering would it be possible to use gpiomem for PWM, then it wouldn't be necessary 
to use root (Kind of interestingly even if you set /dev/mem to 777, it's still not read/writable by user pi).

Cheers
Chris

penfold42, 2018-07-26:
I don’t think so - the linux kernel only exposes the gpio registers

anfractuosity, 2018-07-26:
Ah hmm, I thought it did support SPI?, so I wondered if it could be used for PWM too

penfold42, 2018-07-26:
SPI should be fine as it uses the normal kernel interface /dev/spidev0.0

anfractuosity, 2018-07-27:
I would have liked to have used SPI, but it seems you have to fix the a clock, which lowers the main clock?

I did find some info about PWM with the Pi via - http://www.jumpnowtek.com/rpi/Using-the-Raspberry-Pi-Hardware-PWM-timers.html I'm not sure if that only lets you send a fixed frequency though, so not sure if it'd be usable via this project?

penfold42, 2018-07-27:
Fix it higher ?
Force_turbo used to work for me

anfractuosity, 2018-08-05:
Thanks! I will try SPI mode again sometime, as that seems like it would fix my SD card issue too, as from skimming the code it looks like it doesn't use DMA :)

closed 2018-08-12 by anfractuosity

penfold42, 2018-08-12:
It will end up using dma but it’s the kernel doing it properly not a user space app hittingnthebhardware directly

ningelsohn, 2022-03-24:
Did someone find a solution for using this without root?