jgarff/rpi_ws281x #305: Possible SD card issues with Pi 3B+?

anfractuosity, 2018-07-27:
Hi,

I recently had two pis start to massively hang whilst using my program which embeds your library. (I only recently embedded the ws281x lib and hadn't had any issues like this before, that I noticed).

I got this output from one of them:

```
Jul 27 10:03:08 raspberrypi client[4746]: loop
Jul 27 10:03:09 raspberrypi client[4746]: loop
Jul 27 10:03:10 raspberrypi client[4746]: loop
Jul 27 10:03:10 raspberrypi kernel: INFO: task mmcqd/0:68 blocked for more than 120 seconds.
Jul 27 10:03:10 raspberrypi kernel:       Not tainted 4.14.52-v7+ #1123
Jul 27 10:03:10 raspberrypi kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Jul 27 10:03:10 raspberrypi kernel: mmcqd/0         D    0    68      2 0x00000000
Jul 27 10:03:10 raspberrypi kernel: [<8079be90>] (__schedule) from [<8079c508>] (schedule+0x50/0xa8)
Jul 27 10:03:10 raspberrypi kernel: [<8079c508>] (schedule) from [<8014b1b8>] (io_schedule+0x20/0x40)
Jul 27 10:03:10 raspberrypi kernel: [<8014b1b8>] (io_schedule) from [<8021cc9c>] (wait_on_page_bit+0x110/0x130)
Jul 27 10:03:10 raspberrypi kernel: [<8021cc9c>] (wait_on_page_bit) from [<8027bee0>] (migrate_pages+0x340/0x860)
Jul 27 10:03:10 raspberrypi kernel: [<8027bee0>] (migrate_pages) from [<8022a9d4>] (alloc_contig_range+0x190/0x364)
Jul 27 10:03:10 raspberrypi kernel: [<8022a9d4>] (alloc_contig_range) from [<80285bd0>] (cma_alloc+0xfc/0x288)
Jul 27 10:03:10 raspberrypi kernel: [<80285bd0>] (cma_alloc) from [<8054a948>] (dma_alloc_from_contiguous+0x48/0x4c)
Jul 27 10:03:10 raspberrypi kernel: [<8054a948>] (dma_alloc_from_contiguous) from [<80115578>] (__alloc_from_contiguous+0x40/0xc8)
Jul 27 10:03:10 raspberrypi kernel: [<80115578>] (__alloc_from_contiguous) from [<80115648>] (cma_allocator_alloc+0x48/0x50)
Jul 27 10:03:10 raspberrypi kernel: [<80115648>] (cma_allocator_alloc) from [<80115824>] (__dma_alloc+0x1d4/0x32c)
Jul 27 10:03:10 raspberrypi kernel: [<80115824>] (__dma_alloc) from [<80115a14>] (arm_dma_alloc+0x50/0x58)
Jul 27 10:03:10 raspberrypi kernel: [<80115a14>] (arm_dma_alloc) from [<80272430>] (dma_pool_alloc+0x208/0x23c)
Jul 27 10:03:10 raspberrypi kernel: [<80272430>] (dma_pool_alloc) from [<804f9d3c>] (bcm2835_dma_create_cb_chain+0x1b4/0x214)
Jul 27 10:03:10 raspberrypi kernel: [<804f9d3c>] (bcm2835_dma_create_cb_chain) from [<804fa244>] (bcm2835_dma_prep_slave_sg+0x12c/0x2a8)
Jul 27 10:03:10 raspberrypi kernel: [<804fa244>] (bcm2835_dma_prep_slave_sg) from [<806368b4>] (bcm2835_sdhost_request+0x418/0x5fc)
Jul 27 10:03:10 raspberrypi kernel: [<806368b4>] (bcm2835_sdhost_request) from [<806177a8>] (__mmc_start_request+0x78/0x1a8)
Jul 27 10:03:10 raspberrypi kernel: [<806177a8>] (__mmc_start_request) from [<8061831c>] (mmc_start_request+0x158/0x184)
Jul 27 10:03:10 raspberrypi kernel: [<8061831c>] (mmc_start_request) from [<806185c4>] (mmc_start_areq+0x27c/0x3d0)
Jul 27 10:03:10 raspberrypi kernel: [<806185c4>] (mmc_start_areq) from [<80629b40>] (mmc_blk_issue_rw_rq+0xa8/0x3ac)
Jul 27 10:03:10 raspberrypi kernel: [<80629b40>] (mmc_blk_issue_rw_rq) from [<8062ab14>] (mmc_blk_issue_rq+0x2b4/0x74c)
Jul 27 10:03:10 raspberrypi kernel: [<8062ab14>] (mmc_blk_issue_rq) from [<8062b024>] (mmc_queue_thread+0x78/0x14c)
Jul 27 10:03:10 raspberrypi kernel: [<8062b024>] (mmc_queue_thread) from [<8013d97c>] (kthread+0x13c/0x16c)
Jul 27 10:03:10 raspberrypi kernel: [<8013d97c>] (kthread) from [<8010810c>] (ret_from_fork+0x14/0x28)
Jul 27 10:03:10 raspberrypi kernel: INFO: task jbd2/mmcblk0p2-:69 blocked for more than 120 seconds.
Jul 27 10:03:10 raspberrypi kernel:       Not tainted 4.14.52-v7+ #1123
Jul 27 10:03:10 raspberrypi kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Jul 27 10:03:10 raspberrypi kernel: jbd2/mmcblk0p2- D    0    69      2 0x00000000
Jul 27 10:03:10 raspberrypi kernel: [<8079be90>] (__schedule) from [<8079c508>] (schedule+0x50/0xa8)
Jul 27 10:03:10 raspberrypi kernel: [<8079c508>] (schedule) from [<8014b1b8>] (io_schedule+0x20/0x40)
Jul 27 10:03:10 raspberrypi kernel: [<8014b1b8>] (io_schedule) from [<8079ce78>] (bit_wait_io+0x1c/0x70)
Jul 27 10:03:10 raspberrypi kernel: [<8079ce78>] (bit_wait_io) from [<8079cb08>] (__wait_on_bit+0x94/0xd0)
Jul 27 10:03:10 raspberrypi kernel: [<8079cb08>] (__wait_on_bit) from [<8079cbcc>] (out_of_line_wait_on_bit+0x88/0x90)
Jul 27 10:03:10 raspberrypi kernel: [<8079cbcc>] (out_of_line_wait_on_bit) from [<802c2bc8>] (__wait_on_buffer+0x40/0x44)
Jul 27 10:03:10 raspberrypi kernel: [<802c2bc8>] (__wait_on_buffer) from [<8037adc8>] (jbd2_journal_commit_transaction+0x1340/0x17f0)
Jul 27 10:03:10 raspberrypi kernel: [<8037adc8>] (jbd2_journal_commit_transaction) from [<80380388>] (kjournald2+0x110/0x2c0)
Jul 27 10:03:10 raspberrypi kernel: [<80380388>] (kjournald2) from [<8013d97c>] (kthread+0x13c/0x16c)
Jul 27 10:03:10 raspberrypi kernel: [<8013d97c>] (kthread) from [<8010810c>] (ret_from_fork+0x14/0x28)
Jul 27 10:03:10 raspberrypi kernel: INFO: task kworker/1:0:4608 blocked for more than 120 seconds.
Jul 27 10:03:10 raspberrypi kernel:       Not tainted 4.14.52-v7+ #1123
Jul 27 10:03:10 raspberrypi kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Jul 27 10:03:10 raspberrypi kernel: kworker/1:0     D    0  4608      2 0x00000000
Jul 27 10:03:10 raspberrypi kernel: Workqueue: events_freezable mmc_rescan
Jul 27 10:03:10 raspberrypi kernel: [<8079be90>] (__schedule) from [<8079c508>] (schedule+0x50/0xa8)
Jul 27 10:03:10 raspberrypi kernel: [<8079c508>] (schedule) from [<80617c80>] (__mmc_claim_host+0xb8/0x1cc)
Jul 27 10:03:10 raspberrypi kernel: [<80617c80>] (__mmc_claim_host) from [<80617dc4>] (mmc_get_card+0x30/0x34)
Jul 27 10:03:10 raspberrypi kernel: [<80617dc4>] (mmc_get_card) from [<806209c0>] (mmc_sd_detect+0x20/0x74)
Jul 27 10:03:10 raspberrypi kernel: [<806209c0>] (mmc_sd_detect) from [<8061a68c>] (mmc_rescan+0x1c8/0x394)
Jul 27 10:03:10 raspberrypi kernel: [<8061a68c>] (mmc_rescan) from [<801375a4>] (process_one_work+0x158/0x454)
Jul 27 10:03:10 raspberrypi kernel: [<801375a4>] (process_one_work) from [<80137904>] (worker_thread+0x64/0x5b8)
Jul 27 10:03:10 raspberrypi kernel: [<80137904>] (worker_thread) from [<8013d97c>] (kthread+0x13c/0x16c)
Jul 27 10:03:10 raspberrypi kernel: [<8013d97c>] (kthread) from [<8010810c>] (ret_from_fork+0x14/0x28)
Jul 27 10:03:11 raspberrypi client[4746]: loop
Jul 27 10:03:12 raspberrypi client[4746]: loop
Jul 27 10:03:13 raspberrypi client[4746]: loop
Jul 27 10:03:14 raspberrypi client[4746]: loop
```

I didn't change the DMA from 10, just wondering if you have any ideas?

Cheers
Chris

penfold42, 2018-07-27:
We’ve seen this before.

There doesn’t appear to be a way for a user space program to ask the kernel for a dma channel.
Over time we’ve seen the kernel use different DMA channels. It used to use dma 5 but when this cause issues it was changed to 10.

You could try another dma channel

anfractuosity, 2018-08-12:
Sorry for this issue I could well have been wrong, it seems I just got the same issue, when not using the library so could actually be the SD card itself.

closed 2018-08-12 by anfractuosity