jgarff/rpi_ws281x #307: Installation problem on RPi 3

Simonv92, 2018-07-31:
Hi,
I'm trying to install Neopixel library on my rpi 3 running octopi. I'm having problems when I run sudo python setup.py install as I show you in the photo. I've tryed on two sd card (one of them with fresh install) with same results... 
thank you very much!!

[image]

etanx, 2018-08-01:
This looks like the 'bad zip file' error from thread #290.
Try out hnoesekabel's solution of deleting the old zip file and getting a new one.  Work for you?

Simonv92, 2018-08-02:
Hi,

I've tryed with "sudo pip install rpi_ws281x"and it works 😃thank you!!

A: jgarff/rpi_ws281x
Cc: Simonv92; Author

closed 2018-08-28 by Gadgetoid