jgarff/rpi_ws281x #308: Anyone interested in some dedicated hardware?

JMurph2015, 2018-08-01:
I'm considering making a small-run of custom Pi Hats for the Raspberry Pi Zero that would simplify the hardware connections down to a 5V power in that powers both the Pi and the lights, and four screw terminals for the strip connections.  Anyone interested in this direction (from helping on PCB design to ordering together to increase quantity to outright producing the boards for sale, since we have some commercial guys here too)?

davthomaspilot, 2018-09-15:
I already did something similar, except in my case, I use 12V led strings and have a Pololu step-down regulator for five volts from the 12V bulk power.

Works fine for my purpose.  Cost isn't a big deal--$5 for 10 cards from SEEED, so no need to pool orders.

Gadgetoid, 2022-01-25:
Closing, since this is just a feeeeww years old.

closed 2022-01-25 by Gadgetoid