jgarff/rpi_ws281x #312: Can't make white color or green with WS2812B

Piero87, 2018-08-19:
Hi, i'm trying to color the led stripe of white color and i found this code:

```
import time
from neopixel import *

# LED strip configuration:
LED_COUNT      = 30      # Number of LED pixels.
LED_PIN        = 18      # GPIO pin connected to the pixels (must support PWM!).
LED_FREQ_HZ    = 800000  # LED signal frequency in hertz (usually 800khz)
LED_DMA        = 10      # DMA channel to use for generating signal (try 10)
LED_BRIGHTNESS = 255     # Set to 0 for darkest and 255 for brightest
LED_INVERT     = False   # True to invert the signal (when using NPN transistor level shift)
LED_CHANNEL    = 0

# Define functions which animate LEDs in various ways.
def colorWipe(strip, color, wait_ms=50):
	"""Wipe color across display a pixel at a time."""
	for i in range(strip.numPixels()):
		strip.setPixelColor(i, color)
		strip.show()
		time.sleep(wait_ms/1000.0)

# Main program logic follows:
if __name__ == '__main__':
	# Create NeoPixel object with appropriate configuration.
	strip = Adafruit_NeoPixel(LED_COUNT, LED_PIN, LED_FREQ_HZ, LED_DMA, LED_INVERT, LED_BRIGHTNESS, LED_CHANNEL, LED_STRIP)
	# Intialize the library (must be called once before other functions).
	strip.begin()

	print ('Press Ctrl-C to quit.')
	while True:
		colorWipe(strip, Color(0, 0, 0, 255), 0)  # White wipe
		time.sleep(2)
		colorWipe(strip, Color(255, 255, 255), 0)  # Composite White wipe
		time.sleep(2)
                 colorWipe(strip, Color(0, 0, 255), 0)  # Green wipe
		time.sleep(2)
		colorWipe(strip, Color(255, 255, 255, 255), 0)  # Composite White + White LED wipe
		time.sleep(2)
```

this: `colorWipe(strip, Color(0, 0, 0, 255), 0)  # White wipe`

It's no color the led strip it's seems off

this: `colorWipe(strip, Color(255, 255, 255), 0)  # Composite White wipe`

It's orange

this: `colorWipe(strip, Color(0, 0, 255), 0)  # Green wipe `

it's blue

this: `colorWipe(strip, Color(255, 255, 255, 255), 0)  # Composite White + White LED wipe`

it's no color

can someone tell me how colors works? how can I achieve green or white? or warm white?

Thanks

peos3, 2018-08-25:
The example does not seem correct that way.

It is an RGB strip. So you have three channel intensity values, one for red, one for green and one for blue. Every value is in range between 0 and 255. 0 being off and 255 shining the brightest.

So for Color red you set:
```colorWipe(strip, Color(255, 0, 0))```
for Color green:
```colorWipe(strip, Color(0, 255, 0))```
and for Color blue:
```colorWipe(strip, Color(0, 0, 255))```

To get a solid Color and not a ColorRun you can add a function like this to your script:
```
def colorSolid(strip, color):
	for i in range (strip.numPixels()):
		strip.setPixelColor(i, color)
	strip.show()
```
You don't even need the ```while True``` anymore then.

To get a warm white i would suggest a mixture like this:
```Color(255, 103, 23)```

Piero87, 2018-08-28:
@peos3 Hi, thanks for your answer, take a look here:

https://github.com/jgarff/rpi_ws281x/issues/314