jgarff/rpi_ws281x #313: Driving WS281x with 74HCT08 as a level converter

Mosure, 2018-08-22:
I am trying to use a 74HCT08 AND gate as a voltage converter (3.3v into both inputs of the AND gate while powering the chip with 5V and a common ground.) I connect the output of the AND gate to the LED strip signal wire and it gives the correct color output but only for the first 20ish LEDs. When I connect the RPI's 3.3v signal directly to the strips signal I get mostly correct results (after about 200 LEDs the colors are no longer correct.) I am wondering if this behavior is caused by the output current of the 74HCT08 or if I have overlooked something.

penfold42, 2018-08-23:
Do you have the spare unused inputs tied to either VCC or GND ? They shouldn’t be left floating.

Do you have a decoupling cap across the 74HCT ?

Mosure, 2018-08-23:
I omitted both of these and after adding them it resolved the issue.

closed 2018-08-23 by Mosure