jgarff/rpi_ws281x #319: Incorrect color output - fixed by setting LED count to double what it actually is

garrettrathbone, 2018-08-31:
I've been having weird issues with the LEDs flickering random colors, as well as different intensities of colors producing reproduced but unexpected results - IE Color (0,0,0,127) makes LED 0 off, and LEDs 1 and 2 Green. When fading from Color(0,0,0,0) to Color(0,0,0,255), it starts fading white correctly up until around the 40% mark, then all shut off, then part of the LEDs go green (around the 127 mark), then all shut off again, then go back to all white and work correctly. Same with different intensities of different colors. It always seems that Green will be triggered every 3rd number or so.

Oddly, I fixed it by putting my LED_COUNT as double what it actually is. So if I have 3 LEDs, I state I have 6, and the problem goes away. Is there any rational explanation for this? I have them wired correctly with a 300 uh in the data line, a 5v 15amp power supply, and the grounds connected together.

Side note: I'd been having a hell of a time getting the LEDs to behave correctly before this. First they would flicker randomly and produce random results every other time I ran the script, they worked better when using the 5v directly from the pi, then once I started using a 5v 15amp power supply, it would immediately go bright white the second I even touched the data-line to connect it to the pi. Solved it by getting some insulating coating goop (Super Corona Dope), and coating the portion that I soldered on the back-side of the LEDs. I believe the amperage was so high that it was jumping to the data-line and faking a 1.  So if this isn't a library issue, that's fine, I can keep playing with it. Just seems odd that it's continually reproducible and can be solved by tricking the library.

penfold42, 2018-08-31:
What level shifter are you using ?

VictorioBerra, 2018-09-01:
I am having the exact same problem but i have not been able to fix it.

[shifter](https://www.ebay.com/itm/Adafruit-74AHCT125-Quad-Level-Shifter-3V-to-5V-74AHCT125/253802937551?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649).

[LEDs](https://www.amazon.com/gp/product/B017HAWXF0/ref=oh_aui_detailpage_o07_s00?ie=UTF8&psc=1).

VictorioBerra, 2018-09-01:
Got it! I was NOT SUPPOSED TO GROUND THIS WIRE

[image]

I found this by consulting the ebay pinout and got lucky that I could understand it.

VictorioBerra, 2018-09-01:
Still not perfect... I have one LED green for some reason and some LEDs are purple. I tried to set them all to one solid color.

[image]

penfold42, 2018-09-01:
Pins 4.5.13. 12. 10 and 9 should also be grounded. Unused inputs should always be tied high or low and not left floating.

Try changing the led frequency to 400khz instead of 800khz

VictorioBerra, 2018-09-01:
To clairfy you mean those pins on the level shifter right?

VictorioBerra, 2018-09-01:
Do I have to recompile after editing the main.c to adjust the frequency or can I just change it in the python test scripts?

penfold42, 2018-09-01:
Yes the pins on the 74hct125

Strandtest.py can just be edited and run.

If you are using the c example, you will need to recompile it

garrettrathbone, 2018-09-02:
@penfold42 I'm using the [8-channel bidirectional level shifter from Adafruit](https://www.adafruit.com/product/395).

Like I had said before, it's odd but reproducible, which made me think it could somehow be library related (like maybe the library alters the signal or bits based on how many are listed as in the LED string?). I haven't had a chance to test the LED_COUNT issue since I solidified my layout on a chip and soldered it in yesterday. I ran a test to see if the LEDs lit correctly with the double-LED_COUNT-hack (they did) and then ended up leaving for a few days on vacation.

penfold42, 2018-09-02:
From memory, they don’t like driving long lines. I think they’re designed for chip to chip on a PCB.

Can you try a 74HCT series chip like a 74HCT125