jgarff/rpi_ws281x #323: Is this library software pwm or hardware pwm?

e-3-s, 2018-09-28:
I am using this library to light up my neopixels leds and that works perfect.
I am sending data via a tcp socket. With socket io. 

Without the colorwipe function the data flow is very precise and responsive.
But if I turn on colorwipe with the same data, the led becomes very slow.

The raspberry pi does handle the package very well only very slow.

But when I use a effect the LED strip is very fast.

This means that this library is a bottleneck for the application that I want to use.
Any Idea why?

penfold42, 2018-09-28:
Hardware - but the limiting factor is the led technology.

How many leds do you have ?
Any change to any single led needs a full strip refresh.
24 bits per led, 1.25uS per bit (roughly)
This is 30uS per led so 100 leds takes 3mS to refresh.

Gadgetoid, 2018-10-02:
Most of the bundled examples here illustrate quite elegantly how not to do animation, and couple the animation effect far too tightly to pixel updates.

Normally you should be updating the ws281x pixels continuously at the fastest possible refresh rate for your given strand, and your animation should be timed against wall time and not with `sleep()`.

This means having your code literally look at what time it is (`time.time()` in Python) and using that value to compute the phase of your animation that should be output.

With the animation and framerate decoupled you'll notice that animations appear snappier, but you'll not be able to precisely control them on a frame-by-frame basis. For example a `colorwipe` might light up the strip in blocks of 10 pixels at a time sacrificing accuracy and "smoothness" for speed.

LongLane, 2018-12-23:
You could try DotStars. They are more expensive though.

closed 2022-01-25 by Gadgetoid