jgarff/rpi_ws281x #324: issue building with scons on raspberry pi zero

kevinpanaro, 2018-09-29:
Not much else to say except I was given this output when running scons. Please let me know what other information you may need.

pi@me:~/rpi_ws281x $ scons
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
CC      main.o
/tmp/ccFQiKFd.s: Assembler messages:
/tmp/ccFQiKFd.s:43: Internal error!
Assertion failure in symbol_clone at ../../gas/symbols.c:560.
Please report this bug.
scons: *** [main.o] Error 1
scons: building terminated because of errors.

Zappelphilipp, 2018-10-04:
same here but gives me over 100 errors. a few of them: 
```
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
Version version.h
CC      main.o
In file included from main.c:37:0:
/usr/include/unistd.h: In function 'crypt':
/usr/include/unistd.h:2:6: error: unknown type name '__THROW'
      __THROW __nonnull ((1, 2));
      ^~~~~~~
/usr/include/unistd.h:2:25: error: expected declaration specifiers or '...' before '(' token
      __THROW __nonnull ((1, 2));
                         ^
/usr/include/unistd.h:7:6: error: expected '=', ',', ';', 'asm' or '__attribute__' before '__THROW'
      __THROW __nonnull ((1));
      ^~~~~~~
/usr/include/unistd.h:15:5: error: unknown type name 'ssize_t'
     ssize_t __n) __THROW __nonnull ((1, 2));
     ^~~~~~~
/usr/include/unistd.h:16:2: error: #endif without #if
 #endif
  ^~~~~
/usr/include/unistd.h:47:2: error: #endif without #if
 #endif /* unistd.h  */
  ^~~~~
/usr/include/unistd.h:45:1: error: unknown type name '__END_DECLS'
 __END_DECLS
 ^~~~~~~~~~~
In file included from /usr/include/arm-linux-gnueabihf/sys/types.h:29:0,
                 from main.c:38:
/usr/include/arm-linux-gnueabihf/bits/types.h:30:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'typedef'
 typedef unsigned char __u_char;
 ^~~~~~~
/usr/include/arm-linux-gnueabihf/bits/types.h:31:28: error: storage class specified for parameter '__u_short'
 typedef unsigned short int __u_short;
                            ^~~~~~~~~
/usr/include/arm-linux-gnueabihf/bits/types.h:32:22: error: storage class specified for parameter '__u_int'
 typedef unsigned int __u_int;
                      ^~~~~~~
/usr/include/arm-linux-gnueabihf/bits/types.h:33:27: error: storage class specified for parameter '__u_long'
 typedef unsigned long int __u_long;
                           ^~~~~~~~
/usr/include/arm-linux-gnueabihf/bits/types.h:36:21: error: storage class specified for parameter '__int8_t'
 typedef signed char __int8_t;
                     ^~~~~~~~
/usr/include/arm-linux-gnueabihf/bits/types.h:37:23: error: storage class specified for parameter '__uint8_t'
 typedef unsigned char __uint8_t;
                       ^~~~~~~~~
```
a lot more and at the end:

`char *crypt (const char *__key, const char *__salt)
 
main.c:419:1: error: expected '{' at end of input
 }
 ^
main.c:419:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
At top level:
main.c:31:13: error: 'VERSION' defined but not used [-Werror=unused-variable]
 static char VERSION[] = "XX.YY.ZZ";
             ^~~~~~~
cc1: all warnings being treated as errors
scons: *** [main.o] Error 1
scons: building terminated because of errors.`

kevinpanaro, 2018-10-04:
Hi.

I just logged in to my rpi, and ran scons again, and it compiled no problem...

I think the only thing I did was apt-get update/upgrade.

I'm going to close this issue.

closed 2018-10-04 by kevinpanaro