jgarff/rpi_ws281x #325: Any way to use software PWM ?

6p9hb3ukynd3n8er, 2018-09-29:
Hello,

Is there any way to use software PWM with this library, in order to increase the number of available pins ?

I am using a raspberry pi 3 and I need to control multiple LED strips independently with Python. I have found python libraries that offer software PWM (http://abyz.me.uk/rpi/pigpio/index.html). How could I combine them ?

Alternatively, is there a way to combine PWM, PCM and SPI to control almost 10 strips simultaneously (and they need to be synchronized) ?

Thanks !

Note : I'd rather not use a hat for the rasp.

Gadgetoid, 2018-10-02:
No, and no.

With the possible exception of a from-scratch, bare-metal approach the Pi is simply incapable of generating the precise timings required to produce a valid ws281x signal.

I'd wholly recommend using APA102s strips (Adafruit call those DotStar) since you could switch a whole bank of GPIO pins (one for each strip) and toggle a separate, shared, clock pin to drive a theoretical maximum of 27 independent strips.

If you want to run 10 ws281x strips simultaneously you'll need to do something slightly crazy like- http://spritesmods.com/?art=imx233-ws2811

jgarff, 2018-10-02:
Just be warned, that I think even with a APA102 strip (where you can control the clock line manually), a software PWM frequency using GPIOs is still going to be pretty terrible.  Probably on the order of a few hundred kilohertz.  Depending on the number of LEDs you're driving in each string, you might not be able to achieve a adequate refresh rate using that mechanism.

As for using PWM/PCM/SPI at the same time, the library doesn't support it, but I see no real reason why it couldn't from a hardware standpoint, assuming of course that the dedicated pin functions don't overlap too badly.

JMurph2015, 2018-12-21:
Theoretically, one could create a multiplexer board that converted a PWM signal intended for 1 strip of say 1,000 LEDs to a signal suitable for 10 strips of 100 LEDs.  But this would take custom logic that is probably moderately not worth the trouble.

PS
For my use case, I created a little network server that allows me to remotely stream colors to various networked Raspberry Pi's at the cost of moderately more complexity and no hard guarantees of synchronization (it's a UDP protocol that displays each frame whenever it gets there).