jgarff/rpi_ws281x #326: Cannot run any neopixel code when not root

ramsesvjr, 2018-09-29:
Good Evening! i've been having an issue lately. I have worked with both neopixels and raspberry pis separately for some time now, and i've never had this problem. I can't run my code when its not running on sudo. The problem with this is that the rest of the code uses a module called pyrebase which cannot run on sudo. I would really appreciate anyone who could help me solve. i'll attach my code. I've tried adding 'pi' as a user member of the GPIO group, and nothing happened. i've reinstalled the this software a couple times aswell with no luck.
`
#import Libraries
import RPi.GPIO as GPIO
import time
from pyrebase import pyrebase
from neopixel import*
import argparse

LED_COUNT      = 30      # Number of LED pixels.
LED_PIN        = 18      # GPIO pin connected to the pixels (18 uses PWM!).
#LED_PIN        = 10      # GPIO pin connected to the pixels (10 uses SPI /dev/spidev0.0).
LED_FREQ_HZ    = 800000  # LED signal frequency in hertz (usually 800khz)
LED_DMA        = 10      # DMA channel to use for generating signal (try 10)
LED_BRIGHTNESS = 255     # Set to 0 for darkest and 255 for brightest
LED_INVERT     = False   # True to invert the signal (when using NPN transistor level shift)
LED_CHANNEL    = 0       # set to '1' for GPIOs 13, 19, 41, 45 or 53

def colorWipe(strip, color, wait_ms=50):
    """Wipe color across display a pixel at a time."""
    for i in range(strip.numPixels()):
        strip.setPixelColor(i, color)
        strip.show()
        time.sleep(wait_ms/1000.0)

#Firebase Configuration
config = {
    "apiKey": "\8",
    "authDomain": "7.firebaseapp.com",
    "databaseURL": "https://a.firebaseio.com",
    "storageBucket": "tesa.appspot.com"
}

firebase = pyrebase.initialize_app(config)

#GPIO Setup

#Firebase Database Intialization
db = firebase.database()
    # Create NeoPixel object with appropriate configuration.
strip = Adafruit_NeoPixel(LED_COUNT, LED_PIN, LED_FREQ_HZ, LED_DMA, LED_INVERT, LED_BRIGHTNESS, LED_CHANNEL)
    # Intialize the library (must be called once before other functions).
strip.begin()

#While loop to run until user kills program
while(True):
    #Get value of LED
    led = db.child("led").get()

    #Sort through children of LED(we only have one)
    for user in led.each():
        #Check value of child(which is 'state')
        if(user.val() == "OFF"):
            #If value is off, turn LED off
            colorWipe(strip, Color(0,0,0))
        else:
            #If value is not off(implies it's on), turn LED on
            colorWipe(strip, Color(255,0,0))
        #0.1 Second Delay
        time.sleep(0.1)
`
I removed the firebase credentials from the top for security reasons.``

Gadgetoid, 2018-10-02:
It's not possible to use rpi_ws281x in PWM (or PCM) mode without root. There's no granular way to grant the permissions it needs, since it involves extremely low-level poking of hardware registers.

If you can change which pin/interface used I'm fairly sure (albeit I'm pretty rusty on this at the moment due to other concerns) you can use SPI without requiring root.

The other alternative is to move the low-level access required for rpi_ws281x into a separate application which exposes a socket, or similar interface for your non-root application to control it. This is more or less how every other hardware interface on the Pi works- kernel modules that expose /dev/i2c or /dev/spi interfaces through which the hardware can be controlled.

PandorasFox, 2018-10-09:
You can use [physical pin 19 (gpio 10)](https://pinout.xyz/pinout/pin19_gpio10#) for it to go through SPI. If you're in the `gpio` and `spi` groups then a non-root user can control the strip.

peepsnet, 2018-11-01:
Can you give the commands for this??

PandorasFox, 2018-11-01:
I believe the general command is `sudo usermod -aG gpio,spi <user>`.

You probably shouldn't be doing this under the `pi` user. I would create a new user account that does not have root access, build this library, and then run [strandtest.py](https://github.com/jgarff/rpi_ws281x/blob/master/python/examples/strandtest.py). You should make sure to configure it to use pin 10 like the comment says.

If that does not work, post the groups that user is in, and the exact output you get when running it, ie ```$ python strandtest.py
[output]```

thatneat, 2019-03-06:
@peepsnet @ramsesvjr any luck controlling the LEDs via SPI on `board.D10`? What error message are you getting?

thatneat, 2019-04-12:
I was able to control a neopixel ring without sudo. Here are the steps I took. I didn't exhaustively try with and without all of them so I'm not sure if they're all necessary:

1. physically move LED connection to [BCM10/MOSI/physical pin 19](https://pinout.xyz/pinout/pin19_gpio10#)
2. change pin in code accordingly
3. add pi user to spi and gpio groups: `sudo usermod -aG gpio,spi pi`
4. create `/etc/udev/rules.d/99-com.rules` containing the single line `SUBSYSTEM=="spidev", GROUP="spi"`
5. enable SPI interface through `raspi-config`
6. run the program as pi user without sudo

thatneat, 2019-04-26:
I was able to get it to work without steps 3 and 4 - in the latest raspbian stretch lite, pi is already a member of the necessary groups.

thatneat, 2019-05-27:
OK, quick update - the configuration described above is unreliable on a Raspberry Pi B+ with the latest Raspbian Stretch Lite. The neopixels sometimes change by themselves, and commands sent to them only work ~90% of the time. Maybe there's some other process trying to use the same port?

Here's a related stack exchange question: https://raspberrypi.stackexchange.com/questions/85109/run-rpi-ws281x-without-sudo

AndKe, 2021-12-04:
soon two years later.. has this improved? - maybe the SPI pin is not messed with anymore by some other process?