jgarff/rpi_ws281x #327: Led strip runs one color wipe to green and nothing else

pukeanddie, 2018-10-13:
I am controlling a 12-volt 60pixel neopixel led strip on a Rasberry PI 3 running Stretch. I have it set up through SPI on GPIO 10. I'm controlling the strip directly using 3.3volts as I was not able to get this working with the 5volt level shifter in conjunction with my 12 volt supply for the leds.

I am able to run strandtest.py and if I set the number of LEDS to 60 it will wipe the strip and fill the first color in with green. Then it goes all white and I see the other animation functions being called but nothing on the light changes.

If I reduce the number of LEDS to say 10, I notice that it will fill those 10 pixels green then red and then it goes all white. 

I've monkeyed with a number of settings for frequency, etc with not positive results.

Any suggestions on where to go next to troubleshoot this? My 5-year-old daughter wants to be Alexa for Halloween and I'm running out of time!

pukeanddie, 2018-10-13:
I also tested this on a small 4 pixel 5volt neopixel test strip and I'm getting the same results.

penfold42, 2018-10-13:
You might need to set a fixed core_freq
https://www.raspberrypi.org/documentation/configuration/config-txt/overclocking.md

Ive often done this with force_turbo=1 but you need to beware that this can void warranty if used with other settings

pukeanddie, 2018-10-16:
It's time-based. The strip always goes to white after it runs for ~1000 milliseconds.

pukeanddie, 2018-10-16:
NO @penfold42  you were right. I had originally set the core_freq in the wrong file. Works fine now.

closed 2018-10-16 by pukeanddie