jgarff/rpi_ws281x #342: Raspberry Pi 3 A+ not supported?

LongLane, 2018-12-23:
I am getting:

> File "build/bdist.linux-armv7l/egg/neopixel.py", line 106, in begin
> RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)

It runs fine on a Zero or Pi2

I checked on the Raspberry Pi documentation regarding [HW codes](https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md). In the past, they used sequential codes, whereas it has changed a while ago. The code for the 3A+ is 9020e0 (3B+ a020d3). 

Looking at the code in ws2811.c, `hwver >= 0x010` should be working, but it returns an error. 

I suspect, that the hardware check needs to be brought in line with the new versioning system.

Thanks

penfold42, 2018-12-23:
It needs to be added to https://github.com/jgarff/rpi_ws281x/blob/master/rpihw.c

davthomaspilot, 2019-03-09:
Where can I find the modified rpihw.c for 3A+

davthomaspilot, 2019-03-09:
Sorry, disregard.  Found it.

88weighed, 2019-04-30:
Fixed in this commit:
https://github.com/jgarff/rpi_ws281x/commit/0d99943da134b9db368994c4ccc7a2e1cb109d4d

closed 2019-04-30 by jgarff