jgarff/rpi_ws281x #348: TinkerBoard

RFS96, 2019-01-08:
Hey I`m trying to get this working on the TinkerBoard, I have added the revision in the file but I still get the error. Any ideas?

usefulthink, 2019-01-08:
As far as I know, this library is explicitly built to work with the system-on-chip used by the raspberry-pi  (Broadcom BCM2835 and so on) only. It is very unlikely, that the SoC used in the TinkerBoard (Rockchip RK3288) would be compatible with that.

Main reason is that the library directly accesses some aspects of the processor via memory-mapped I/O (DMA, PWM and PCM) and even if other processors provide the same features they are unlikely to use the exact same memory-adresses.

RFS96, 2019-01-08:
Thanks for the reply. Any suggestions on how to get it working considering It has a pwm pin just like the rpi?

usefulthink, 2019-01-08:
That's gotta be very hard to do, pretty close to rewriting the entire library. You will at least need to understand all the ways this library interacts with the hardware via mmio and study the processor-reference for the RK3288 (if you can find them) to find the corresponding features, hoping they are even available (which isn't granted).

penfold42, 2019-01-08:
SPI should work

usefulthink, 2019-01-09:
In that case, I stand corrected. 

For a better understanding, this is because `linux/spi/spidev.h` abstracts hardware-differences away?

penfold42, 2019-01-09:
Yes - let the OS do the heavy lifting.
This does however require that the OS and SPI drivers support DMA to ensure the timing

SteMMo33, 2019-11-29:
Hi all, I'm interested to port the library on Tinker, too.
Or, did anyone find an alternative ?

SteMMo33, 2019-12-18:
Hi all,
I'm able to run on Tinkerboard.
I selected the SPI channel and, tailing the source code, I'm able to compile library and test application.
I see the data on the scope but the signal is not suitable for the strip: usually after a block of data 32us long, the data presents some delay/hole and the chips do not understand the sequence.

penfold42, 2019-12-18:
SPI support assumes the Linux kernel will use DMA to ensure no gaps in the data.

If the tinker board SPI interface or the kernel driver doesn’t do DMA (and DMA with a long buffer) this won’t work.

SteMMo33, 2019-12-18:
I'm using a Linaro distribution or a Armbian OS.