jgarff/rpi_ws281x #349: Raspberry PI 3 / WS2812B flickering

Urmel11, 2019-01-13:
Hi,

I am using a "Raspberry Pi 3 Model B Plus Rev 1.3" to drive a WS2812B LED strip (110 LEDs at the moment).
As you can see in the [video](https://www.youtube.com/watch?v=4rAIXHtioZM), the strip is flickering randomly. It is even flickering, when I try to drive 11 LEDs.

I am using a level shifter (74AHCT125N). The power supply provides 5V and 10A.
As mentioned in the ReadMe, I already blacklisted "snd_bcm2835" and added the missing lines to config.txt.
I am using [this](https://learn.adafruit.com/assets/19658) wiring.

Does anybody have an idea what is wrong?

hilts50, 2019-01-19:
I am having the same issue.  Not sure what would be causing the flickering.

klaasnicolaas, 2019-08-30:
Are you using `python` or `python3`?